heishamon/HeishaMon #448: Conflictive actions in home assistant

drsugranes, 2023-12-28:
Hi.
First of all thanks all of you for the developement and the board.
I use it with Home Assistant and, when I change some values, it start a loop with the automations or return to the original value. In example:
1- I change the quiet level at HA
2- HA start the automation Set Heatmode and sends the topic (topic1) with the new value
3- Meanwhile (up to 5 seconds), Heishamon board publish the value (topic2).
4- HA detecs the topic (topic2) and run Set Heatmode Selector
5- Heishamon board process the topic1

This is the reason I modify the yaml to stop the automations selector during 5 seconds. This is the file I use. Hope it should be useful.

################################################################
## Packages / Heishamon
################################################################

################################################################
## Changelog
################################################################
##
## 1.0.0 (08-01-2020)
##   Added
##   - Initial version
##
## 1.1.0 (10-01-2020)
##   Added
##   - Added power switch and Fan RPM sensors
##
## 1.2.0 (17-05-2020)
##   Fixed
##   - New/changed topics
##
## 1.3.0 (22-06-2022)
##   Compatibility with Home Assistant 2022.6 MQTT configuration format
##
## 1.4.0 (18-09-2022)
##   Added
##   - Split heatshift into separate Z1 and Z2
##
## 1.5.0 (16-10-2023)
##   Added
##   - Cool modes
##   - All modes COP calculations
##   Removed
##   - S0 energy calculations (we are already calculating it in sensors)
## 1.5.1 (18-10-2023)
##   Fixed
##   - Power templates when values not available
## 1.5.2 (28-12-2023)
##   Fixed
##   - Wait to enable selectors automations

# Automations #
###############
automation:
  #Fetches quiet level from status and sets the selector accordingly
  - alias: Set quiet level selector
    trigger:
      platform: mqtt
      topic: "panasonic_heat_pump/main/Quiet_Mode_Level"
    action:
      service: input_select.select_option
      data_template:
        entity_id: input_select.heishamon_quietmode
        option: >-
          {%- if trigger.payload == "0" -%}
            Off
          {%- else -%}
            {{ trigger.payload }}
          {%- endif -%}

[... 857 lines cut ...]

  #Calculation Energy consumed (kWh) - DHW
  - platform: integration
    source: sensor.aquarea_dhw_power_consumed
    name: Aquarea DHW Energy Consumed
    unit_prefix: k
    method: left
  #Calculation Energy consumed (kWh) - Heat
  - platform: integration
    source: sensor.aquarea_heat_power_consumed
    name: Aquarea Heat Energy Consumed
    unit_prefix: k
    method: left
  #Calculation Energy consumed (kWh) - Cool
  - platform: integration
    source: sensor.aquarea_cool_power_consumed
    name: Aquarea Cool Energy Consumed
    unit_prefix: k
    method: left
  #Calculation Energy consumed (kWh) - Total
  - platform: integration
    source: sensor.heishamon_power_consumed
    name: Aquarea Energy Consumed
    unit_prefix: k
    method: left

geduxas, 2023-12-28:
Yes it's known problems. Thanks for sharing your suggestions, but please make proper push request for changes in yaml.

drsugranes, 2023-12-29:
Ok, thankyou.

closed 2023-12-29 by drsugranes