heishamon/HeishaMon #449: Add WH-ADC0916H9E8

Bohoo69, 2023-12-30:
Hello,

Could you add WH-ADC0916H9E8 (internal unit 3 phase from 9 to 16 kW depending on external unit, mine is a 12 Kw  WH-UX12HE8)

Received 203 bytes data
data: 71 C8 01 10 55 55 52 49 00 5A 00 00 00 00 00 00 00 00 00 00 19 15 11 55 16 9D 55 05 05 00 00 00 
data: 00 00 00 00 00 00 80 80 80 80 B4 76 01 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B7 A3 7B 8F B7 A3 7B 8F 96 82 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 03 3D 7B C1 06 00 00 00 00 00 00 00 00 55 F5 55 21 A9 15 55 05 1B 12 66 00 00 00 00 00 00 00 
data: 00 E2 CF 0C 74 09 12 D0 0E 95 05 91 00 9F 89 91 91 32 32 A9 A9 32 32 32 80 A9 91 86 91 92 89 88 
data: 89 89 91 31 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 DB 03 00 F6 01 00 03 00 00 02 00 00 0D 
data: 02 01 01 01 01 01 01 00 00 00 FE 
Checksum and header received ok!

Many thanks in advance

King regards

Ludo

artlov, 2024-01-03:
I have posted request for the same model, but 16 kW a couple mounths ago:
https://github.com/Egyras/HeishaMon/issues/407

Bohoo69, 2024-01-04:
Many thanks !

nrahbek, 2024-01-09:
> I'll add both soon

Could you add the following as well:

32 D4 0B 89 84 73 90 0C 86 84 | Monoblock | WH-MXC16J9E8 | Monoblock | 16 | 3ph | T-CAP

Thanks in advance :)

HighlyCompressedAir, 2024-09-11:
All three were added, this issue can be closed.

Bohoo69, 2024-09-12:
Agreed i'ts ok now !