heishamon/HeishaMon #450: WH-SQC12H9E8 External Control Contact

BlueCase, 2023-12-31:
Hi,
is it possible to get the state about the external contact switch?

MiG-41, 2023-12-31:
Yes,  in byte 110 bit7&8 givess actual status , you could extract it from topic panasonic_heat_pump/raw/data
If you want to have it as one of TOPs , maybe @IgorYbema will add it...

BlueCase, 2023-12-31:
@IgorYbema would that possible?

BlueCase, 2024-01-21:
@MiG-41 Do you know or have an example how to get this working with homeassistent

MiG-41, 2024-01-21:
I can give you simple flow for NodeRED.
..

BlueCase, 2024-09-19:
@MiG-41 sorry for the lat reply but I had other topics with more priority :). thanks for the offer. can you provide the flow?
@IgorYbema Any Update. Would it possible to add?

MiG-41, 2024-09-20:
Somethink like this:

[{"id":"3fb91e04a87c11b7","type":"mqtt in","z":"29ae07782cca960c","name":"","topic":"panasonic_heat_pump/raw/data","qos":"2","datatype":"auto","broker":"70855f4b.303b1","nl":false,"rap":false,"inputs":0,"x":190,"y":180,"wires":[["0dd4a9e922de38b0","49b0545a45cc3209"]]},{"id":"0dd4a9e922de38b0","type":"function","z":"29ae07782cca960c","name":"110","func":"\nmsg.payload = Number((msg.payload[110] & 0b11111111) >> 7 );\n\nreturn msg ;","outputs":1,"timeout":"","noerr":0,"initialize":"","finalize":"","libs":[],"x":410,"y":180,"wires":[["fd839e51f775cad0"]]},{"id":"fd839e51f775cad0","type":"mqtt out","z":"29ae07782cca960c","name":"","topic":"panasonic_heat_pump/raw/External switch","qos":"","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"70855f4b.303b1","x":670,"y":180,"wires":[]},{"id":"49b0545a45cc3209","type":"function","z":"29ae07782cca960c","name":"110","func":"\nmsg.payload = Number((msg.payload[110] & 0b00111111) >> 5 );\n\nreturn msg ;","outputs":1,"timeout":"","noerr":0,"initialize":"","finalize":"","libs":[],"x":410,"y":240,"wires":[["97609334b7603dc8"]]},{"id":"97609334b7603dc8","type":"mqtt out","z":"29ae07782cca960c","name":"","topic":"panasonic_heat_pump/raw/CoolHeat","qos":"","retain":"false","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"70855f4b.303b1","x":650,"y":240,"wires":[]},{"id":"70855f4b.303b1","type":"mqtt-broker","name":"","broker":"192.168.1.90","port":"1883","clientid":"","autoConnect":true,"usetls":false,"verifyservercert":true,"protocolVersion":"4","keepalive":"15","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","birthMsg":{},"closeTopic":"","closePayload":"","closeMsg":{},"willTopic":"","willQos":"0","willPayload":"","willMsg":{},"userProps":"","sessionExpiry":""}]

Import it to node red. It will read topic panasonic_heat_pump/raw/data ( so you need to Enable in Heishamon settings "Debug log to MQTT topic from start" ) , and it will set give a result in topic panasonic_heat_pump/raw/External switch (and also for edu purpose panasonic_heat_pump/raw/CoolHeat )....

BTW , this Bit for Cool Heat Switch may be not true , since it gives me no results when Cooling... So please veryfy if , External Switch gives proper results....

BlueCase, 2024-09-27:
@MiG-41 Thanks for your help. Will try it :)

closed 2024-11-03 by github-actions[bot]