heishamon/HeishaMon #455: bivalent connection

meavric, 2024-01-07:
Hi, is posible to get state bivalent connection from WH-SDC0709J3E5 9 1ph HP ?

meavric, 2024-01-08:
Hi,
I have a heater connected to the buffer as a bivalent source (it turns on at -15 degrees C). Is it possible to derive the fact that this heater is turned on from the heat pump?

MiG-41, 2024-01-08:
For a real state of bivalent relay (and maybe defrost state if Bivalent is not active in J and higher models) , is responsible Bit1&2 in byte 112... Please ask @IgorYbema ,if he is willing to add this information in TOPs :)

boon244, 2024-01-10:
Hello, I would also be interested in this...who can I buy a coffee for?

Ioniq88, 2024-01-13:
Is it possible to integrate the Parameters for bivalent  connection to Heishamon?

- Enable/disable bivalent connection
- set outdoor temperature for turn ON Bivalent connectin
- Bivalent mode Alternative, Parallel, Advanced Parallel
- External Pump ON/Off 

This would be grate

Best regards Uli

geduxas, 2024-10-14:
is this same as #377 ?

closed 2024-11-21 by github-actions[bot]