heishamon/HeishaMon #457: Feature request - control direct heat / compensation curve setting

taugusti, 2024-01-09:
I use Heishamon to control the floor heating and hot water in my summerhouse via home assistant and the hass plugin.
it all works like a bliss!

I have a "vacation mode" setting that changes the setpoint and mode of the heatpump.

I would like to reduce the on/off cycles in non-vacation mode. I would like to do this by opening all floor heating circuits and control the temperature by setting the direct heat setting. It does not have to be accurate - just efficient.

For this purpose, I would like to change from compensation curve to direct heat.
I have found a sensor that reflects the current configuration - but not a setting to change it.

So - the feature request is to add the ability to change the mode between direct heat and compensation curve control.

Thanks!

geduxas, 2024-01-09:
Could you point me out how direct heat could reduce on/off count?

taugusti, 2024-01-09:
> Could you point me out how direct heat could reduce on/off count?

Sure :-)
by averaging out the heat request, with open loops and a low forward temperature, I expect that the pump will deliver a constant power instead of starting and stopping due to too low delta temp on forward and return lines.
At least that is the theory :-)

geduxas, 2024-01-09:
No, hetapump will stop if supplied heat will increase by 3 degrees over setpoint..

jllussa, 2024-01-11:
@geduxas Can you share the URL to download the full manual?

taugusti, 2024-01-11:
@geduxas 
I can see that the start/stops I experience are directly coupled to the periods where the floor heating closes all valves.
When only the bypass valve is open, the heat pump cannot disipate the energy and Tout - Tset will increase above 3 degC -> turn off.

If all valves are always open abd i control Tout to be as low as possible while still achieving the desired 11 degC out-of-season heating, the frequency of not being able to disipate the heat will become much more infrequent.

So yes - I am pretty sure it will work.

The reason I would like to be able to configure the direct heat / compensation curve is to be able to switch to compensation curve "in season" where I would like the system to run as today.

McMagellan, 2024-01-13:
Why don't you change the heating curve parameters directly.

For example: Heating curve operation:
TOP29 @Z1_Heat_Curve_Target_High_Temp = 38
TOP30 @Z1_Heat_Curve_Target_Low_Temp = 28

If you now change both parameters to a value of your choice, you will always get the desired value as the flow temperature, regardless of the heating curve calculation.

With the command:
192.168.178.XXX/command?SetCurves={zone1:{heat:{target:{high:26,low:26}}}}
You always get a fixed 26°C target temperature.

To switch back to the heating curve, send the original values back:
192.168.178.XXX/command?SetCurves={zone1:{heat:{target:{high:38,low:28}}}}

Unfortunately there is no write command in Rules for SetCurves, which is why this can only be done externally.

stumbaumr, 2024-01-16:
Related to https://github.com/Egyras/HeishaMon/issues/311

CurlyMoo, 2024-04-16:
Setting the curves inside the rules should now be possible with the latest code in this PR:
https://github.com/IgorYbema/HeishaMon/pull/121

It should still be tested though.

Setting the curves is then done like this:
````ruby
@SetCurves = concat('{"zone1":{"heat":{"target":"high":', @Z1_Heat_Curve_Target_High_Temp, ',"low":', @Z1_Heat_Curve_Target_Low_Temp, '}}}}');
```

CurlyMoo, 2024-05-18:
@Egyras This one can be closed.

closed 2024-05-19 by taugusti