heishamon/HeishaMon #458: Received 0 bytes data

SimonBard, 2024-01-13:
For some days, Heishamon is no more sending current data. It sends only outdated data via MQTT. The heatpump is now again controlled by the control panel of panasonic. The values given that way are not reflected by heishamon. 

I now updated to 3.2.3, but still have the same issue. 

I get a log in MQTT, but what exactly does it say?

```
sent bytes: 111 including checksum value: 18 
Previous read data attempt failed due to timeout!
 Received 0 bytes data
```

Can that be a problem with the cable? Or does it mean the physical connection is ok, but its a software problem? 
MQTT is at least powered by the heatpump electronic. I can also reach the "Heisha monitor" (see screenshot)
[image]

IgorYbema, 2024-01-13:
Did you try power cycle the heatpump? So pull all the power from the heatpump so it resets.
We have, for now, only seen this in firmware v3.2.0 with the optional pcb installed. A power cycle resolves the issue.

Of course the cable can also be broken but that doesn't happen suddenly.

The message you see is just that heishamon is sending it's query for the data but it doesn't receive the answer back.

PGnl, 2024-01-13:
Make sure you first switch off the heatpump on the controller and wait a minut before taking the power off.

SimonBard, 2024-01-13:
Thank you for the prompt reply. I switched the power off and on and it works again. Sorry for the stupid issue.

JockelBln, 2024-01-13:
@SimonBard Can you please close this issue if everything is ok again?
Thanks

Coowly, 2024-07-20:
I have, sinds previous update (2-3 weeks back) (dont know if it was H.A. or heishamon, they both updated at the same time) exactly the same issue. Unfortunately a power cycle doesn't work. At start I got onetime correct values after that i have the exact same line;

sent bytes: 111 including checksum value: 18 
Previous read data attempt failed due to timeout!
 Received 0 bytes data

closed 2024-09-12 by Egyras