heishamon/HeishaMon #460: Issue with TOP 13, 115 and 66

Bohoo69, 2024-01-21:
Hello,

I have seen some issues with the following topic 
- TOP13 Main Schedule State stays always at 0 even if Scheduled state is activated (showed on Heat Pump screen)
- TOP66 Low water pressure is always at 0. I find the attached schematic, slighty different than the one and there is no low pressure sensor on Aquerea ? Is this for other models than Aquerea T-Cap ?

As seen in issue resquest #445, water pressure is negative -0.02 for me

Many thanks in advance

Ludovic

MiG-41, 2024-01-21:
You dont have a water pressure sensor in your HP , so you recive sch data. This sensor is onlynin new K/L models.
 Also TOP66 is not for water , it is for gaś pressure ,and also your HP don't have it (and to be honest don' know with have . )
Regarding TOP13 i can't say anything.

Bohoo69, 2024-01-22:
Ok thanks for the return I remove this two items... Perhaps TOP66 is present on multisplit or otherequipements

MiG-41, 2024-01-22:
Witch sensor you belive it is Copressor Tank Temp ?

Bohoo69, 2024-01-22:
I beleive - not sure - that this is top 55. What is surprising is when HP is down TOP55 is higher than 5 to 10°C from external temperature near the room temp where is situated internal parts of the HP. On the maintenance documents of Panasonic, the IPM temperature is located outside. I hesitate between compressor temp and frequency convertor driving the compressor.

[image]

geduxas, 2024-01-22:
IPM, it's measuring temperature of power module (inverter)

MiG-41, 2024-01-22:
https://americas.fujielectric.com/faqwd/what-does-ipm-mean/

Bohoo69, 2024-01-23:
Ok thanks for the information, this is the inverter temp and not compressor temp.

closed 2024-09-12 by Egyras