heishamon/HeishaMon #463: Resend homeassistant discovery topic

SimonBard, 2024-01-23:
Is there a chance to resend the mqtt discovery topic?

I have rebooted heishamon and visited "http://192.168.178.45/togglelog". It sends the mqtt data, but homeassistant need a config topic, which I cannot see in the mosquitto logs. 

How can I trigger resending the config messages?

geduxas, 2024-01-23:
Heishamon don't have it.

SimonBard, 2024-01-23:
@geduxas: Thanks for the reply, which I do not understand. 
Is heishamon not sending config topics at all?
Or is there no resend function? Because the resend function needs to be triggered at some point automatically. I just wanted to know when exactly so I am able to trigger is myself.

[image]

geduxas, 2024-01-23:
Sorry for too short answer, yes heishamon don't send any discovery/config topic. It's sends only it's data. 
As i understand from screenshot your referring to heishamon integration. So this issue should be rised in integration repository. 

Try to look in mqtt server via mqttexplorer or similar tool, maybe topic is not updated at all.

closed 2024-09-12 by Egyras