heishamon/HeishaMon #465: limit compressor On/off

dimaregabriele, 2024-01-26:
Hi everyone
I bought and installed the device and connected it to my Panasonic heat pump. Since this heat pump is a little oversized for my system, it often does a lot of on/off. I have already set the silence level to 3. Do you think it is possible to avoid these continuous shutdowns via software by telling it not to switch off for a certain amount of time?

geduxas, 2024-01-26:
Here is condition's for compressor control from heatpump service manual, maybe it will help for your case
[image]

McMagellan, 2024-01-27:
Controlling the silence level (Quiet Mode Level) from Heishamon is entirely possible with a small rules program.
This allows the compressor frequency to be throttled to a small extent. The main problem with clocking is the relatively high minimum power at the lowest compressor speed of 19 Hz. A sensible temporary use of the Quiet Mode Level is therefore only to slow down the starting procedure.

geduxas, 2024-01-27:
> Controlling the silence level (Quiet Mode Level) from Heishamon is entirely possible with a small rules program. This allows the compressor frequency to be throttled to a small extent. The main [...]

There is also demand control.. which is possible to emulate with heishamon, but also i am unable to handle it lower 19-20Hz, so i think it's lowest possible value to run compressor. 

Also with quaet mode i noticed more slower spin of air blades..

naseweiss99, 2024-02-04:
There are at least two ways to calm the Jeishas etc.

1) demand control - limit to minimum (dec 50 with my T-Cap) 0 is full blast.:-)
2) use a rule to permanently adjust the target temperatur in that way that you are near the point where the heatpump shuts down . This will bring it ito the max. reduction of input power.

example :

this rule will start with -4° target temp shift and end up with max. 5° shift
Iset in the Display the start temp to -4°

---------
on System#Boot then
 
 

  #Takt = 0;
  
settimer(1,30);
end

on timer=1 then 

   
  if @Heat_Power_Consumption > 0 then
    if @Z1_Heat_Request_Temp < 5 then
      if  @Main_Outlet_Temp >= @Z1_Water_Target_Temp + 2 then
             @SetZ1HeatRequestTemperature = @Z1_Heat_Request_Temp + 1;
      end
     end  
   end 
    if @Heat_Power_Consumption == 0 then
	            if @Pump_Flow < 20 then
				    #Takt = #Takt + 1;
					if #Takt > 2 then
						if @Z1_Heat_Request_Temp != -4 then 
	  
							@SetZ1HeatRequestTemperature = -4;
							#Takt = 0;
						end
				    end	  
		        end 
    end
   settimer(1,30);
end

McMagellan, 2024-02-04:
I solved the frequent clocking in a similar way with a very complex rule. I increase the Z1 Heat Request temperature by 1 degree in case the compressor is about to switch off. See the purple line in the middle in the graph. 

I currently allow it for 2 degrees so that the upper hysteresis point is raised by 2 degrees. Without this rule, the compressor would switch off when the first degree was set. I compensate for the resulting overheating of the rooms with an integrally controlled blocking time after the end of the cycle within a specified time period. 

I only set the silencelevel 2 for 2.5 minutes immediately after starting. The goal is to keep the compressor frequency at a minimum for as long as possible (see blue line below).

According to the design implemented here, there can never be more than 9 cycles per day. The rule is very complex and is currently in the development phase. I have a 5KW Panasonic Monoblock J Serie.

[image]

geduxas, 2024-02-04:
@McMagellan could you share your rules?

McMagellan, 2024-02-05:
Last night was very chaotic and I was very annoyed with Heishamon and Rules. Unfortunately, it turned out that Rules is unreliable and not secure. Therefore, I will now implement my program in iobroker Blockly.

At 8 p.m. the rule crashed and then the HP ran without rules at the default until 1 a.m. Here you can see exactly how the rule, when it works, lowers the clock frequency. There were no interventions on TOP27 during this time, see the purple line in the middle in the graph.

At 5 a.m. there was a komplet power outage and as a result the rule no longer started. The Heishamon system date is January 1st, 1970 with a dummy time and is no longer updated. In the morning there were 3 more rules crashes. I was able to record one of them in the console and you could see that an endless timer had simply stopped running.

With these experiences, I will stop further development in Rules because there is no availability. I am now putting the experience I gained into Blockly.

Attached you will find my rule, which I have slimmed down so that only the TOP27 correction and the blocking period run. It may well be that she is running better now. You are welcome to test it at your own risk to see how the beats change. The worst thing that can happen is that the WP is switched off at the beginning of a blocking period and the Ruel no longer switches it on.

Do you also work with Blockly?

[image]

The prerequisite for the rule is the heating curve mode and no room thermostat and only the heating circuit Z1.

[Projekt 12.1 small Test.txt](https://github.com/Egyras/HeishaMon/files/14169422/Projekt.12.1.small.Test.txt)

geduxas, 2024-02-05:
Thanks for sharing, no i am not familiar with blockly, i am HomeAssistant/node-red kind of user, also i am not using any rules, from heishamon i perfere as bridge interface.. yes it would be best if it could run locally, also this kind of rules, compiled directly in its firmware could run more stable and better....

naseweiss99, 2024-02-06:
@geduxas the less complex rule on top is mostly doing what is required to get long runs without on/off of the heatpump.

However there are two things, which should be **changed or configurable in Heishamon**.

1) Due to the fact that rules and/or Heishamon crash from time to time, rules should e.g. restart automatically again 5min after reboot.
    In current code they are disabled after a reboot due to a crash. 

2) the NTP call should be done twice after a boot/reboot - 2nd time for example 5mins after boot. 
    Background is that after a power shortage, brown out what ever, the Network routers providing internet connection are not 
   ready immediately but need some minutes to sync etc.

ciao

martin31821, 2024-06-14:
I'm in a very similar situation, we have the WH-MXC12J9E8 and I'm wondering if someone ever has tried to reduce the minimum frequency from 19 hertz down to 15/12hz? 

I know a few Daikin Users where the Compressor actually runs all the way down to 8/12hz, whereas the Aquareas never go below 19, **except** when they are starting up: mine is starting up very smoothly from 0 all the way to 35hz, and you can even hear the compressor spinning, so *from a technical point of view* it should be possible.

mi-hol, 2024-09-12:
@dimaregabriele did version 3.8 fix your issue? 
In case yes, please close it, otherwise comment.

geduxas, 2024-09-12:
> @dimaregabriele did version 3.8 fix your issue? In case yes, please close it, otherwise comment.

It's always open and discussed topic how to prolong compressor work, and lower on/off cicles.. this should move to discussion topic. Most of users comes with own experiment's and tried to find out why most efficient way..

closed 2024-10-22 by github-actions[bot]