heishamon/HeishaMon #473: settings lock

manuhassio1, 2024-03-12:
The home screen correctly indicates the heat pump data, while if I go into the wifi configuration screen it remains stuck with:Please wait, loading saved settings...
I tried resetting the board and turning off the power but it remains blocked.

manuhassio1, 2024-03-14:
[image]

geduxas, 2024-03-14:
Which version you're running?

manuhassio1, 2024-03-15:
Version 3.2.3.
yesterday I also tried to replace the firmware with an older one but nothing changed, the home screen gives me the aquarea values ​​while setting remains stuck waiting.

manuhassio1, 2024-03-27:
I reset everything and reinstalled with a usb-ttl cable from version 3.2 to 3.2.3 and with all of them, once the mqtt section was set and the configuration was saved, the heishamon network reappears.

closed 2024-09-12 by Egyras

henkedk, 2024-09-13:
@Egyras 
Is there a fix?
I just upgrade to 3.8 - my settings page is stuck on
"Please wait, loading saved settings..."

geduxas, 2024-09-13:
@henkedk try to open in incognito mode and check.. it could be something cashed in your browser.

henkedk, 2024-09-14:
> @henkedk try to open in incognito mode and check.. it could be something cashed in your browser.

Same thing - I think you got a bug due to special chars in the MQTT password - I got both " and ! in mine.

geduxas, 2024-09-14:
@henkedk could you somehow share your password with @IgorYbema to check it out.. if it's reproducible it maybe possible to fix, or at least warn other about it..

henkedk, 2024-09-15:
> @henkedk could you somehow share your password with @IgorYbema to check it out.. if it's reproducible it maybe possible to fix, or at least warn other about it..

I've sent him an email

henkedk, 2024-09-17:
> @henkedk could you somehow share your password with @IgorYbema to check it out.. if it's reproducible it maybe possible to fix, or at least warn other about it..

I didn't hear anything from him, mind checking in with him to see if he got my email? :)