heishamon/HeishaMon #477: MQTT - Operating_Mode_State - How to deactivate

Skol6, 2024-04-11:
Hi Folks,

How can I deactivate the pump over MQTT?

Current operation modes are:
- 0=Heat only
- 1=Cool only
- 2=Auto(Heat)
- 3=DHW only
- 4=Heat+DHW
- 5=Cool+DHW
- 6=Auto(Heat)+DHW
- 7=Auto(Cool)
- 8=Auto(Cool)+DHW)

In winter operation I switch always between 0, 3 and 4.
But how to perform summer operation: 3 and ??

Current workarround is to set SetHeatingOffOutdoorTemp to 5°C to prevent heating to start. And stay with 0, 3 and 4.
Do you have any other ideas?

Regards

geduxas, 2024-04-11:
If put fully in standby use power on/off

IgorYbema, 2024-04-11:
Yes, in summer mode you just keep mode on 3 and switch heatpump off if no DHW is needed (SetHeatpump=0) and turn it on again to heat up the DHW (SetHeatpump=1).
Depening if you need cooling, this logic changes again (just like winter, but then with the cool modes)

geduxas, 2024-04-11:
@IgorYbema is there anything specific to keep pump off, and switch on only to heat dhw?

IgorYbema, 2024-04-11:
Not sure what you mean, sorry

geduxas, 2024-04-11:
> Not sure what you mean, sorry

You suggested to turn pump off, and turin it on only when dhw reheat is needed :) so thought is there any point for that?

Skol6, 2024-04-11:
Well I have tried to switch the heatpump completely off. Of course this is working, BUT
- It beeps every time
- I cannot lock at the current DHW temperature

IgorYbema, 2024-04-11:
> > Not sure what you mean, sorry
> 
> You suggested to turn pump off, and turin it on only when dhw reheat is needed :) so thought is there any point for that?

For example if you don't want the DHW to heat up on expensive tariff.

IgorYbema, 2024-04-11:
> Well I have tried to switch the heatpump completely off. Of course this is working, BUT
> 
> * It beeps every time
> * I cannot lock at the current DHW temperature

It beeps? I never had that. Don't remember turning of any beeper sound also.

Checking the DHW temp can be done on your own automation ofcourse :)

Skol6, 2024-04-15:
Thanks for the informations. Than I stick to my idea using SetHeatingOffOutdoorTemp instead of beeping and shoing no information.

closed 2024-04-15 by Skol6

IgorYbema, 2024-04-15:
Maybe I disabled the beeper. Don't know anymore :-) But I use this all the time.

geduxas, 2024-04-15:
> Maybe I disabled the beeper. Don't know anymore :-) But I use this all the time.

I'm curious also about beeping :D never had any problem's with that. Does it emit beep powering on/off ? Maybe if you send couple "off" command's it's beep's on every?

MiG-41, 2024-04-15:
In user menu there is option like Touch panel sound , when you tourn it off ,then also during SetHeatpump there is no beep ( but touchpad also not beeps ,when you press any key...) . 
Next issue is the background light of controler. It also turn on for set time during SetHeatpump . You can also permanently turn off backlight , but still....

@geduxas  , it beeps only one time , so sending permanently same state in some intervals to Setheatpump will generate only one first beep on change and backlight on.... :)