heishamon/HeishaMon #478: Control of DWH heating element

Skol6, 2024-04-15:
Hi Folks,

In case I have a lot of energy delivered from my PV system, I want to activately switch on the external electric DHW heater.
Do you know how this is possible with MQTT commands?

Regards!

IgorYbema, 2024-04-15:
No that is impossible. However there are other options:
- just set the DHW setpoint temperature higher than current+delta and the heatpump will run a DHW
- disconnect the electric heater from the heatpump and connect it to a switchable relay (like sonoff) and control it that way (then also enable that relay if heatpump enables the heater which you can find in one of the heishamon values)

kgiedrius, 2024-04-18:
but it is possible to turn on/off DWH and ROOM heaters from controll panel. Why it is not possible with integration ?

MiG-41, 2024-04-18:
No , it is not possible to switch them , it is possible only to deactivate them ( so they are not turning on during heating - generally) , but you can't just turn on/off them , for this it is HP with decides .
( there is a possibility to turn on manually ,but this is service menu , HP have to be switched off ).

MiG-41, 2024-04-18:
I use for that porpuse Shelly1PM . This turns on/off DHW heater , and at the same time to contact it is connected HP , so when HP want's to turn on heater , shelly do it as well.

IgorYbema, 2024-04-18:
Even if you do a 'force dhw heat' on the controller panel (or using heishamon command) this will not exactly do what you want (only use the heater element) and will not always start immediatly.
Using the trick as @MiG-41 described is a good solution and fits my 2nd option in my post above.

HighlyCompressedAir, 2024-09-12:
Large version has two 230V/10A relays and can achieve what the author wants. Small version requires additional hardware. This issue can be closed.

closed 2024-10-22 by github-actions[bot]