heishamon/HeishaMon #479: Command SetCompressorState seemingly doen't work

krispek, 2024-04-23:
The commmand ```SetCompressorState``` doesn't work. At least, that's what I think. 

My goal is to disable the internal heater as much as possible, at least for defrosts. The only reason that I would like to use the internal heater, is the sterilization mode. I make use of floor heating, which offers more than enough heat to defrost the heat pump. 

Based on the manual, the commmand ```SetCompressorState``` might be able to do so. It looks like a software version of the setting ```Backup heater (Use/Do not use Backup heater)``` in the custom menu, and this forces the internal heater to be turned off while defrosting. However, it does not work in the way I expect it to work. The internal heater is still being used to defrost the heat pump. 

What I do, is use the command ```SetCompressorState``` with the value ```1```. This does not result in any feedback whatsoever. This is expected, since the manual states ```There is NO pop up screen like External Switch when this External Compressor Switch is OFF.```

There's a couple of possible explanations for this:
- This command does not affect the use of the internal heater for defrosting
- Somehow this command needs an additional argument (the manual mentions Dip switch Pin 3, which cannot be set afaik)
- Dip switch Pin 3 is set to off, which means that the heat source (that is the heat pump itself) should not work. I verified that this is not the case. :) 

Did anybody else try to use the command SetCompressorState? And what can I expect from this command?

geduxas, 2024-04-23:
This command is from expansion PCB, do you have one? Have you turned on PCB emuliation, also this requires to enable external PCB

krispek, 2024-04-23:
Yes, I did this. Other commands (SetDemandControl) do work fine.

closed 2024-10-17 by github-actions[bot]

krispek, 2025-04-27:
Thanks for this, it now works. I have a J-series, and given the manual I think that the dip switch is on the same place. The default setting of the switch is compressor, since I can now turn on and off the compressor. :) 

Next winter I will check whether I can prevent the heater to turn on during defrosts, which is the goal.

geduxas, 2025-04-27:
Just turn off heater in service mode to prevent it.. if you absolutely shure heater is not required in your setup.

krispek, 2025-04-27:
I need the heater for the legionella protection cycle. :)