heishamon/HeishaMon #481: Problem uploading a new rule.

McMagellan, 2024-05-04:
Sometimes it happens that I cannot upload a rule. 
The Rules window freezes. Apparently the routine cannot find the correct end of the file and the console then shows a message like this: 

Sat May 4 09:29:31 2024 (329783): Reaching end of read. Readlen: 473, Totallen: 533

An excerpt from the console log shows the following:
[image]

I have a permanent problem with this attached file. 
[concat Testrule V1.txt](https://github.com/Egyras/HeishaMon/files/15209852/concat.Testrule.V1.txt)

If I change the length of the rule by inserting lines, the upload works. 
This rule only runs on the current beta version #485 (thanks to Igor and CurlyMoo) and includes the new commands @SetCurves, concat and print. The upload process also fails, for example in older versions like 3.2.3 .
.
.
.
.
In addition, I sometimes have the problem that with a longer rule the upload process is aborted with the following message: Too large read. (see middle of picture)
[image]

IgorYbema, 2024-05-04:
Yes there is a bug which causes it to not read it when it is a specific length. This bug is fixed but not made to production yet.

McMagellan, 2024-05-04:
Thanks for the quick reply. This error has been bothering me for a long time. Good to hear that a fix is ​​coming. Igor asked me not to post any unrelated topics in the "Update rule library" thread, which is why I'm putting it here and hope it's in the right place.

IgorYbema, 2024-05-04:
That was CurlyMo asking you that, not me :)

IgorYbema, 2024-05-04:
Wait, I didn't read it correctly. You also get the error on the alpha builds? I just tried and indeed the webloading seems to hang using this ruleset.

McMagellan, 2024-05-04:
I'm currently using version Alpha-8aa33db, #485
I just wrote a long-term test program with concat, print and SetCurves with and without numeric values and got the “Too large read” error when uploading. Added two blank lines at the end and it can be loaded. Both loading errors in #485 have definitely not been eliminated.

[image]

IgorYbema, 2024-05-05:
I believe I found it. Took me a few hours. Can you test it? 
https://github.com/IgorYbema/HeishaMon/pull/121#issuecomment-2094701566

McMagellan, 2024-05-05:
I switched to version Alpha-f9b90fd #487.
now you have to be strong. Unfortunately, during the first test as described above, I had the same error as described above. After rebooting and reopening the window and saving with an empty window, the next attempt to load the rule worked. Several subsequent loading processes went smoothly. 
The error can therefore only be reproduced by chance and is probably dependent on the state of the Heishamon internal memory usage.

You asked about the test rule that yesterday said "Too large read!" caused.
After a 20-hour test operation, I reloaded the rule today in the old firmware status #485 and the error no longer occurred and can therefore not be reliably reproduced.
You can see that he was there yesterday in the picture above at the time stamp, May 4th 4:11 p.m.

Attached is the test rule. Again, I think it depends on Heishamon's current internal memory state.

[concat Testrule V2a.txt](https://github.com/Egyras/HeishaMon/files/15213304/concat.Testrule.V2a.txt)

IgorYbema, 2024-05-05:
Ok thanks. Yes I believe that there is still another bug in there which is hard to reproduce but this is much better already.

McMagellan, 2024-05-07:
Today I had the error again when loading a test rule with "Too large read" in version Alpha-afd39aa, #494.
After I reopened the crashed rules window and reloaded the same rule, it worked. this error cannot be reproduced.

[image]

CurlyMoo, 2024-06-10:
This issue seems to be fixed.

closed 2024-06-10 by McMagellan