heishamon/HeishaMon #482: Restarts of Heishamon V3.2.3

Keisha0815, 2024-05-06:
Hallo team,
I´ve a panasonic heatpump K-line and I´ve two problems with the heishamon (Connected to the extra board (CZ-NS5P):

Firmware V3.2.3 is installed. It restarts often , i can not see the reason for it. The WiFi seems to be stable: more then 90%. The power connection with a 5V Phone Connector doesn´t help. 

The second problem: The XTOP-entrys , a time they are visible, now they are unvisible. Why?

Can you help me, please?

Thanks a lot, Keisha0815

Keisha0815, 2024-05-06:
Hallo team, it seems the first problem is solved: a WiFi-Repeater next to Heishamon . About 13 hours without restart... 
but where are the XTop entries? Any ideas?

IgorYbema, 2024-05-07:
Do you have a K or L series? Only XTOP for these ones.
And it is detected on boot of heishamon if your heatpump supports it. Try to reboot heishamon once again please.

Keisha0815, 2024-05-07:
hi Igor, 
I´ve rebootet heishamon again and now I can the XTOP´s .
Thanks a lot,
Keisha0815 :-)

Keisha0815, 2024-06-11:
Hi Igor,
yesterday at 23:55 the heishamon lost the connection to the heatpump. After a restart there are no XTop entries. Can I do something else as put on the reboot button? 
thanks a lot.
Armin

Keisha0815, 2024-06-12:
Hi Igor, after another reboot I ve seen TOP 92 shows me 	WH-MDC05H3E5
but its a Model K. 🤔
Is the new bigger Ethernet-Version of Heishamon better for the new K/L-Line?
Best regards, Armin

Keisha0815, 2024-06-12:
Hi Igor,
after a power cycle my heishamon runs, the XTOP´s are back.
What I´ve done before it lost the connection:
A few hours earlier i´ve set "Emulate optional PCB" in Settings. I figured out the [SetDemandControl=53] Option,
it seems the option only work in Emulate PCB Mode. The heishamon is connected to the CZ-NS4P so this is a problem?
I would trie to slow down the heatpump because its too fast it begins to cycle ...
Best regards, Armin

closed 2024-07-11 by Keisha0815