heishamon/HeishaMon #483: Time synchronization fails.

McMagellan, 2024-05-09:
Since last year the variables %hour and %minute have been implemented in rules. The values ​​correspond to the same time as the timestamp in the console.

I have a power outage about every 6 weeks. When heishamon restarts (uptime starts new) the date is set to January 1st. 1970 and the time restarts at 1:00. Obviously the synchronization fails and no further attempt is made. I suspect that at the time of synchronization my Fritzbox (WLAN to Internet connection) had not yet established a working connection. This usually takes me about 6 minutes. 
If I reboot Heishamon later, the time is correct again.

Would it be possible to carry out further synchronization attempts at appropriate intervals if, for example, the year internally is still set to 1970?

[image]

RichieB2B, 2024-05-10:
Even for systems that are able to do successful time synchronization at boot, it is good practice repeat the synchronization regularly. Clocks tend to drift which is why time synchronization was invented.

CurlyMoo, 2024-05-18:
https://github.com/IgorYbema/HeishaMon/pull/134

stumbaumr, 2024-09-06:
@McMagellan, you could close this now, right?

closed 2024-09-06 by McMagellan