heishamon/HeishaMon #484: REQ: option to disable the WiFi hotspot

RichieB2B, 2024-05-18:
Now that Heisamon supports wired Ethernet, it should be possible to disable the WiFi hotspot completely.
This setting should also prevent Heisamon to automatically present the passwordless hotspot when it cannot connect to the configured WiFi network.

Alternative: add the option to set a password for the hotspot. This way we can still use it as fallback but it will be more secure.

IgorYbema, 2024-05-18:
Yes, already working on this

schellevis, 2024-07-04:
Will this also be available for Heishamon Small? In my case, the Heishamon setup hotspot is accessible from the street. I don't have Ethernet on my Heishamon, but the trade-off of having to reset the device with the onboard button for increased security seems acceptable to me.

HighlyCompressedAir, 2024-09-12:
Feature implemented #517, issue can be closed.

closed 2024-10-22 by github-actions[bot]