heishamon/HeishaMon #487: Communication Error H74

Skol6, 2024-05-23:
Hi Folks,

Heishamon was running for several weeks without any issues.
during the last days I got multiple times commincation error H74.

Not nowing what could be the problem I have performed a complete power cycle of my aquarea heat pump.
Afterwards everything is running normal again.

Can someone confirm this issue?

Keisha0815, 2024-06-12:
Hi Skol6,
my Heishamon yesterday losts its connection to the heatpump too. Where have you read the error number?
I´ve tried to reboot but it does not run. After a complete power cycle it would connect again and now it works until ...?
Good Luck, Armin

Keisha0815, 2024-06-12:
Hi, i‘ve not seen any errors in the heishamon menu. Have you installed the CZ-NS5P?
Wifi-Problems?

Skol6, 2024-06-30:
I am not using the  CZ-NS5P.

IgorYbema, 2024-06-30:
Not using the CZ-NS5P but you are getting H74? Then you most be using the optional pcb emulation in heishamon and enable the optional pcb in the heatpump.
The emulation is not perfect. If heishamon fails/get delayed the communication stops and causing the H74. This can be caused by so many reason but mainly network related as heishamon is a single threaded device not able to handle network delays together with communication to the heatpump at the same time.

Skol6, 2024-07-05:
Yes I am using the PCB emulation to control the heat pump. 
H74 is really only for communication between main PCB and optional PCB - that is simulated by heishamon.

So bug must be something to do with the Heishamon Software.

IgorYbema, 2024-07-05:
Yes but as I said the emulation is not perfect. The H74 pops up if heishamon is a bit too busy and therefore not delivering the messages in time. It helps if you keep heishamon not to active like lowering the timers.

closed 2024-10-17 by github-actions[bot]