heishamon/HeishaMon #488: Feature request: Comment in rules (v5 Hardware)

blb4github, 2024-05-25:
With the new Hardware (v5, based on ESP12), the space for rules is increased and rules do require less space (if I'm correct).
It would be convenient to have the possibility to add comment to the rules to document the rules in the rules set itself.

McMagellan, 2024-05-29:
This is already partly possible with beta version 3.5.
In order to be able to implement the command @SetCurves in Rules, (needs JSON string) the internal handling of strings was integrated. 
For example, up to 255 characters can be assigned to a local or global variable. (#Test1 = 'Good morning together';)
This is then displayed in the list of supplied variables in the lfd console.

Additionally, the "Print" command was introduced which displays the text in the console. 
This command can also be used as a note for documentation in the lfd rules code. (print('### Standby overflow ###');)
The disadvantage is that this text is output in the console during runtime which would not be necessary here.

Out of the box:
I have already made the suggestion (#121) to implement a Rem command as a modified print command without any output.
As a further idea, I also suggested implementing a command such as SendUrl = 'Text', which can send the attached string to the web server. This could be used, for example, to control the relay on a Shelly. I could also have disruptions or daily statistics sent to my cell phone via string to a WhatsApp Bot. The text can be up to 255 characters long.
Maybe it can be implemented after all because it would fit well with the introduction of strings.

blb4github, 2024-05-29:
Thanks, useful! I have now implement in my rules a command `print('#Time: ', #Time);`
I have experimented with the string variable (add in a lot of lines with #com1 = 'xxxxxxxx xxxxx xxxxx xxx'; but this gives some reboots. I'll check further.

McMagellan, 2024-05-30:
@blb4github:
I have your command "`print('#Time: ', #Time);`" installed in my SetCurves test rule and it runs without any problems.
I'm using version Alpha-baddb94, build #531 on Heishamon V4

In this example you have used a numeric value in addition to the text. 
This application makes no sense for the Rem application you requested.

For the sake of completeness, in such a case the command "concat( )" is used to insert a number into a text. 
The print command is a variant of the concat command and therefore already includes the number-text conversion.

Example for concat: @SetCurves = concat('{zone1:{heat:{target:{high:',#HKTempUp,',low:',#HKTempDown,'}}}}');

CurlyMoo, 2024-06-10:
> I think if it were easy to implement it would definitely already exist.

It is relatively easy. Altough it's not a feature for the rules library. As in, the rules library is not going to support it. It is possible to implement a feature that filters all comments before the rules are provided to the rules library.

closed 2024-10-16 by github-actions[bot]