heishamon/HeishaMon #489: Panasonic Model K WH-ADC0309K3E5 + WH-UDZ09KE5 not recognized in Heishamon V5 with firmware V3.5

ZonPV, 2024-05-30:
This evening I connected the new Heishamon V5 with firmware V 3.3.5 to my heatpump. 
In the HeishaMon it says that Topic 92: Model is Unknown.

The hardware I have is a WH-ADC0309K3E5 + WH-UDZ09KE5

I was wondering a bit as in Tweakers.net [https://gathering.tweakers.net/forum/list_messages/2102536?data%5Bfilter_keywords%5D=model%20k](url) I got the method to detect the right data by the command http://IP-adres/debug. I gave the following dump:

-- heatpump data --
[000]: 71 C8 01 10 55 55 62 49 00 59 00 16 00 00 00 00 00 00 00 00 59 15 13 55 56 1D 55 55 56 19 56 00 
[032]: 00 00 00 00 00 00 96 9C A3 80 B4 71 71 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
[064]: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A0 A3 7B 8F A3 A3 7B 8F 98 85 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
[096]: 8A 0D 5B 7C BF 0B 7E 7C 1F 7C 7E 00 00 00 55 55 55 21 87 15 55 05 0A 11 66 21 51 15 01 56 80 80 
[128]: 80 E2 D5 0D 99 94 02 D6 0D 68 95 92 00 B0 90 A1 9D 32 32 A0 A3 40 32 32 96 A0 A2 94 92 9A 8F 00 
[160]: 90 00 00 3F 01 01 01 00 00 22 00 01 01 01 01 79 01 01 01 1E 02 00 0C 02 00 01 00 00 08 00 00 0A 
[192]: 01 00 00 00 00 00 00 03 80 41 8E 
-- extra data --
[000]: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 11 00 01 00 01 00 01 00 01 00 01 00 01 00 00 00 01 00 
[032]: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[064]: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[096]: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[128]: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[160]: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[192]: 00 00 00 00 00 00 00 00 00 00 13 

Maybe that info was not completed or there might be another reason? Never mind. I ask this question again and in this forum. I hope the model I own will be in the next firmware.

Thanks for all the good work.

mi-hol, 2024-09-06:
@ZonPV was this issue fixed in new version 3.7?

mi-hol, 2024-09-06:
[image]

ZonPV, 2024-09-06:
I am still on original V 3.5.. 

Your screenshot notes 2 different powervalues, 7 and 9 kW... Seems to be a mistake..🤔
A while ago I added my hardware in the list in the NodeRed flow of @edterbak.

ZonPV, 2024-09-06:
@mi-hol 
Updated to V3.7, the latest release of this morning, was successfull. But TOP92 is still unknown.... 
/
[image]

[image]

No idea how to correct it.. 
Can anyone confirm that the model number is listed in TOP92 of the Heishamon? Thnx in advance

mi-hol, 2024-09-06:
> No idea how to correct it..

From commit history of https://github.com/Egyras/HeishaMon/commits/master/HeatPumpType.md and [the code implementing it](https://github.com/Egyras/HeishaMon/blob/0202dd58c1aa51ffc9561a3af40853ec923a886f/HeishaMon/decode.h#L41) I'd assume @IgorYbema is updating it.
Maybe [Readme ](https://github.com/Egyras/HeishaMon?tab=readme-ov-file#using-the-software) should be extended with a new sub topic "How to: Add a new heat pump type"

geduxas, 2024-09-06:
@ZonPV are you really on latest version? From version :alpha*** it's sounds like it's nightly build, all stable versions have version number.

@mi-hol for adding new models it's required to open issue with correct dump and post whole hexdump of packet, it will be implemented later. Also it's good to check @IgorYbema repo, becose all latest development going there.

mi-hol, 2024-09-06:
> From version :alpha*** it's sounds like it's nightly build,

@geduxas please see my finding in #513

geduxas, 2024-09-06:
> > From version :alpha*** it's sounds like it's nightly build,
> 
> @geduxas please see my finding in #513

It's better to post in @IgorYbema repo.

mi-hol, 2024-09-06:
> It's better to post in @IgorYbema repo.

@geduxas IgorYbema's repo is just a fork, posting there is not possible
[image]

geduxas, 2024-09-06:
Sorry overlooked, though it's pr :) yes, all issues here . Contributions and pr' butter there

mi-hol, 2024-09-06:
> > From version :alpha*** it's sounds like it's nightly build,
> 
> @geduxas please see my finding in #513

but in the meantime I downloaded and opened for example https://github.com/Egyras/HeishaMon/blob/master/binaries/model-type-large/HeishaMon.ino.esp32-v3.7.bin in an editor

There is indeed a proper version number

[image]

@ZonPV maybe an error happened when downloading and flashing the latest release

mi-hol, 2024-09-06:
@ZonPV suggest to wait until https://github.com/Egyras/HeishaMon/pull/514 is closed!

ZonPV, 2024-09-06:
@mi-hol 
Thnx for your effort this problem. I assume you own an K-model with 9kW too? 
No hurry as I asked for adding this model a long time ago on Tweakers.. 
It is a bit unclear where to put these issues about Heishamon to me.

@geduxas 
What is the difference between a nightly build and build with version number? I had in mind that the nightly build is the latest verion.. And I read earlier ap post from @thehog on Tweakers forum that after an update they saw that text..

[image]

But yes indeed, I am afraid he is not here on this forum..

IgorYbema, 2024-09-06:
Hi,
Your model isn't in the list yet (https://github.com/Egyras/HeishaMon/blob/master/HeatPumpType.md).
It really doesn't matter if it is 'unknown' or not. It works fine. The model number is just hard coded info and doesn't make thing better or worse.

But I'll add it so it will be in there in a later release.

The alpha builds are test builds for specific changes. Only recommended for the ones who are helping debugging code.

ZonPV, 2024-09-07:
Thnx for info. It feels as fancy when a persons toy is recognized in a list. I understand that it is fake..🙂

Never the less I added my model of the Panasonic in the node-red flow of @edterbak to make the dashboard more complete.

I will have patience. Thnx for all the good work!

ZonPV, 2024-09-08:
@mi-hol 
This morning I read on Tweakers Heishamon forum that Igor updated the firmware to V 3.8 and added my model in the list. 
Just did the update to the Heishamon and this question is solved.  So I close this issue. 
Thnx to @mi-hol, @IgorYbema  and @geduxas

mi-hol, 2024-09-08:
@ZonPV thanks. 
Now to really  "close" this issue, please press "close issue" button 

see https://docs.github.com/en/issues/tracking-your-work-with-issues/closing-an-issue

closed 2024-09-08 by ZonPV