heishamon/HeishaMon #491: Invalid json sent to MQTT

kamaradclimber, 2024-06-09:
From time to time it seems invalid json messages are sent to MQTT. 

For instance:
`{"uptime":2253526185,"voltage":3.33,"free memory":70,"free heap":28224,"wifi":100,"mqtt reconnects":7368,"total reads":450255,"good reads":450244,"bad crc reads0]panasonic_heat_pump/logSun Jun  9 18:03:34 2024 (2253531506): Requesting new pana`

or

`{"uptime0]panasonic_heat_pump/logSun Jun  9 08:03:46 2024 (2217544209): Requesting new panasonic data0mpanasonic_heat_pump/logSun Jun  9 08:03:47 2024 (2217544756): sent bytes: 111 including checksum value: 18 0panasonic_heat_pump/LWTOnli`

it seems to happen only on `panasonic_heat_pump/stats` topic.

kamaradclimber, 2024-07-18:
@Egyras would that issue ring a bell? 

[Here](https://p.datadoghq.eu/sb/833f0246-221f-11ee-a842-da7ad0900005-7c4a4853f78310af0a7b3ed1d30ee006?refresh_mode=sliding&theme=light&from_ts=1718731799217&to_ts=1721323799217&live=true) is a count of those events on my installation (sadly I can't share the logs but you would see what I wrote in my first message, repeated over and over).

kamaradclimber, 2024-07-18:
@geduxas and @IgorYbema might also have ideas on this

geduxas, 2024-07-18:
I had noticed something like this in past.. but it was mqtt server issues..

In other cases, it could lead to memory issues..

kamaradclimber, 2024-10-09:
I don’t see traces of this anymore, let’s close for now.

kamaradclimber, 2024-10-09:
Actually I can see traces of this in the last 15 days (around 60 occurrences)

geduxas, 2024-10-09:
> Actually I can see traces of this in the last 15 days (around 60 occurrences)

Is it possible to run some wireshark/tcpdump in mqtt server and capture network traffic from heishamon for a while.. it's important to get network level data..

kamaradclimber, 2024-10-10:
It seems to happen quite infrequently, I’ve setup a tcpdump of mqtt traffic from heishamon and will be alerted when the issue reoccur.

geduxas, 2024-10-10:
Personally I suspect some TCP stack buffer overflow.. or coruption.. strange things happens while trying to update frimware while listening in console window.. so i think something simmilar could lead to corruption of MQTT packet..  almost same happens for other's in rules...  i had in past some problems with connection stability, but everything went away after i stoped accessing web page of heishamon at all. 

https://github.com/IgorYbema/HeishaMon/pull/138#issuecomment-2368390265
https://github.com/kamaradclimber/heishamon-homeassistant/issues/248#issuecomment-2341764320

closed 2024-11-21 by github-actions[bot]