heishamon/HeishaMon #493: Where my Variables in the Console? (Heishamon V5, SW 3.5)

Binifada1956, 2024-06-10:
Today, installed the new Heishamon. I was wondering while i don't see my Global Variables in the Console. Now no Control what my Rules doing!

stumbaumr, 2024-06-11:
V3.5 still has some bugs.
See
https://github.com/IgorYbema/HeishaMon/pull/121
and 
https://github.com/IgorYbema/HeishaMon/pull/133

Binifada1956, 2024-06-12:
I tested a diversion today with print(...). Here I can output the variables in the console, but is that the way it's meant to be?
In addition, the print command requires a lot of resources.

stumbaumr, 2024-06-13:
Without showing your code it is really hard to say what goes wrong...

stumbaumr, 2024-06-18:
I would recommend that you try the build from here https://github.com/IgorYbema/HeishaMon/actions/runs/9553126605 and check to see if that fulfills your requirements.

If you are still missing functionality get involved [here](https://github.com/IgorYbema/HeishaMon/pull/133)...

Binifada1956, 2024-08-21:
ok, Variables are in Cconsole again 👍

closed 2024-08-21 by Binifada1956