heishamon/HeishaMon #496: Formatting a float number in rules.

McMagellan, 2024-06-12:
To calculate the COP, I would like to display a value with one decimal place. (See $COPIntern in image)

[image]

The calculation is 2800 / 460 and it should display 6.1.
What should the formula look like?

stumbaumr, 2024-06-13:
Without a built-in round()-function this is a rather long piece of coding...

McMagellan, 2024-06-13:
I also tried it with `round( )`.
To demonstrate, I wrote a small test rule to show the problem.
The formula is to multiply the internal amperage of the inverter by 230V and add a flat rate of 50 watts for the fan. Then you get the power input of the refrigeration circuit, by which the amount of heat is then divided. This is how I calculate the COP, which should be displayed as a number with only one decimal place.

The test rule carries out the calculation based on your formula with different values ​​for ampere, which is also a decimal number.
    `$BK1COP = round(#Leistung / ((#Ampere * 230) + 50) * 10) / 10;`
    `$BK1COP = round(3800 / ((2.0 * 230) + 50) * 10) / 10;`
Here is the result with different ampere values: (2.0 to 2.7)

[image]

A new problem was noticed when parsing the test rule.
I use Version Alpha-0776a05, build #537 on Heishamon V4.

ERROR: cannot compute * with a left char value

[image]

Here is the test rule. If the last two lines in Timer1 are removed the rule is accepted. 
The malfunction is reproducible for me.

[Round Testrule failed.txt](https://github.com/user-attachments/files/15828313/Round.Testrule.failed.txt)

stumbaumr, 2024-06-13:
Adjusted your rule and let it run...
```
on System#Boot then
  settimer(1,10);
  #Ampere = 2.0;
  #Leistung = 3800;
end

on timer=1 then
  settimer(1,1);
  $COP = round(#Leistung / ((#Ampere * 230) + 50) * 10) / 10;
  #Ampere = #Ampere + 0.1;
end
```
[HeishaMon-round-float-errors.txt](https://github.com/user-attachments/files/15829859/HeishaMon-round-float-errors.txt)

Looks like a float addition error to me...

stumbaumr, 2024-06-14:
8 Watts idle,
Waterpump energy usage as a function of pump duty,
Fan energy usage as a function of rpm
and compressor as a function of the compressor current.

And still using an external energy meter and an external heat meter is more accurate...

McMagellan, 2024-06-14:
Here is a comparison showing how the Rules `round( )` command works when calculating with a decimal number.
You can see the course of a real heating cycle. 

I will not use the round command for this in the future and am satisfied with the normal result.
However, it is critical because, for example, you expect a value of 4.9 for an if command and a value of 4.89999 is returned. 
Then the if command doesn't work. (if xx == 4.9 then)

This COP display is completely sufficient for me. I would like to have the opportunity to spontaneously view the current COP.
It's just a cosmetic mistake not to have a display with only one decimal place.

[image]

closed 2024-10-09 by McMagellan