heishamon/HeishaMon #497: [Feature Request] Extend rules with MQTT subscriptions

martin31821, 2024-06-14:
I'd like to propose an addition to the rule engine in heishamon, namely the possibility to subscribe to MQTT topics and run rules when a message is received on the topic, as well as the possibility to react to connection lost/reconnection events.

Possible usecases might be:
- Adapting to variable electricity tariffs locally on the heatpump, including a smart fallback if the tariffs aren't available
- Adapting to weather forecast
- Dynamically adapting to PV leftovers
- user-defined integrations e.g. for homeassistant etc.

I'm imagining a rule like this:

```basic 
on System#Boot then 
 
  # mqttSubscribe takes two parameters, first the topic to subscribe to 
  # and a target callback to invoke when a message is subscribed 
  mqttSubscribe("electricity/currentprice", 10)  
 
end 

# When a message is received that's been registered with callback number 10
on MQTT#10 then 
  # MQTT topic for current message is stored in @MQTT_Topic 
  if @MQTT_Topic == "electricity/currentprice" then 
    # Value shall be a float / number ? 
    # Available in @MQTT_Value 
    if @MQTT_Value < 0.28 && @DHW_TEMP < 45 then 
      @SetDHW_Target_Temp = 55 
    elseif @MQTT_Value > 0.50
      @SetDHW_Target_Temp = 35
    end
  end 
end 

# React to connection loss from broker
on MQTT#lost then 
  @SetDHW_Target_Temp = 45 
end 
```

What do you think of such an extension?

martin31821, 2024-06-25:
I do agree with you on the long term logging, although I have mine running in grafana/prometheus/mimir. 

I was thinking to run more logic directly on device in order to have a sane fallback in case any of the more complex rules fails, thus also the handling of `MQTT#lost`.

henrikssn, 2024-09-27:
I have a similar use case: Allow changing global variables from external sources.

The rationale would be to have some "settings", which are not time critical for the operation of the heat pump but I want to be able to change these from MQTT or HTTP.

If we had MQTT subscriptions as rules, then I could just set the global variables as needed in the MQTT subscription rule.

closed 2024-11-03 by github-actions[bot]