heishamon/HeishaMon #498: Lots of bad header reads, heatpump goes offline

klausj1, 2024-06-30:
I have connected my Heishamon (bought from IgorYbema, Version: 3.5) instead of the Panasonic module directly to the heatpump. Heishamon is connected to the Home Assistant via MQTT.

The heatpump is a brand new L-Series installed end of april.

Since some time, I get lots of bad reads:

[image]

Here is am example protocol of the Heishamon console log:

```
Sun Jun 30 10:03:01 2024 (2046398907): Heishamon stats: Uptime: 23 days 16 hours 26 minutes 38 seconds ## Free memory: 63% ## Heap fragmentation: 14% ## Max free block: 21152 bytes ## Free heap: 24536 bytes ## Wifi: 100% (RSSI: -45) ## Mqtt reconnects: 11946 ## Correct data: 97.06% Rules active: 0
Sun Jun 30 10:03:01 2024 (2046398926): Requesting new panasonic data
Sun Jun 30 10:03:01 2024 (2046398929): sent bytes: 111 including checksum value: 18 
Sun Jun 30 10:03:01 2024 (2046398936): Requesting new panasonic extra data
Sun Jun 30 10:03:01 2024 (2046398943): Already sending data. Buffering this send request
Sun Jun 30 10:03:05 2024 (2046402974): Previous read data attempt failed due to timeout!
Sun Jun 30 10:03:05 2024 (2046402976): Received 0 bytes data
Sun Jun 30 10:03:05 2024 (2046402978): Sending command from buffer
Sun Jun 30 10:03:05 2024 (2046402984): sent bytes: 111 including checksum value: 1 
Sun Jun 30 10:03:05 2024 (2046402999): Received 203 bytes data
Sun Jun 30 10:03:05 2024 (2046403001): Checksum and header received ok!
Sun Jun 30 10:03:05 2024 (2046403007): received TOP67 Compressor_Current: 1.0
Sun Jun 30 10:03:05 2024 (2046403014): received TOP115 Water_Pressure: 2.56
Sun Jun 30 10:03:06 2024 (2046403908): Heishamon stats: Uptime: 23 days 16 hours 26 minutes 43 seconds ## Free memory: 64% ## Heap fragmentation: 15% ## Max free block: 21152 bytes ## Free heap: 24672 bytes ## Wifi: 100% (RSSI: -45) ## Mqtt reconnects: 11946 ## Correct data: 97.06% Rules active: 0
Sun Jun 30 10:03:06 2024 (2046403928): Requesting new panasonic data
Sun Jun 30 10:03:06 2024 (2046403930): sent bytes: 111 including checksum value: 18 
Sun Jun 30 10:03:06 2024 (2046403937): Requesting new panasonic extra data
Sun Jun 30 10:03:06 2024 (2046403945): Already sending data. Buffering this send request
Sun Jun 30 10:03:07 2024 (2046404523): Received bad header. Ignoring this data!
[deleted 54 identical rows]
Sun Jun 30 10:03:07 2024 (2046404955): Received bad header. Ignoring this data!
Sun Jun 30 10:03:07 2024 (2046404984): Received more data than header suggests! Ignoring this as this is bad data.
Sun Jun 30 10:03:08 2024 (2046405939): Previous read data attempt failed due to timeout!
Sun Jun 30 10:03:08 2024 (2046405941): Received 0 bytes data
Sun Jun 30 10:03:08 2024 (2046405942): Sending command from buffer
Sun Jun 30 10:03:08 2024 (2046405948): sent bytes: 111 including checksum value: 1 
Sun Jun 30 10:03:09 2024 (2046407099): Received 203 bytes data
Sun Jun 30 10:03:09 2024 (2046407101): Checksum and header received ok!
Sun Jun 30 10:03:11 2024 (2046408909): Heishamon stats: Uptime: 23 days 16 hours 26 minutes 48 seconds ## Free memory: 60% ## Heap fragmentation: 12% ## Max free block: 20504 bytes ## Free heap: 23176 bytes ## Wifi: 100% (RSSI: -45) ## Mqtt reconnects: 11946 ## Correct data: 97.06% Rules active: 0

```
But the Heishamon says that 97% are OK (last row above)

Any ideas?

Apart from that, Heishamon is working fine (as far as I can see, because my heatpump does nothing, as the hot water is produced with solar thermal energy; at least the sensor values are available and make sense. And there are no reboots)

klausj1, 2024-06-30:
Maybe its two different problems  nad reads and going offline:

It seems the heatpump goes offline when it looses the MQTT connection to the HA:

[image]

And the corresponding Heishamon Log:

`
Sun Jun 30 10:04:01 2024 (2046458919): Heishamon stats: Uptime: 23 days 16 hours 27 minutes 38 seconds ## Free memory: 59% ## Heap fragmentation: 8% ## Max free block: 21152 bytes ## Free heap: 22936 bytes ## Wifi: 100% (RSSI: -46) ## Mqtt reconnects: 11946 ## Correct data: 97.06% Rules active: 0
Sun Jun 30 10:04:01 2024 (2046458937): Requesting new panasonic data
Sun Jun 30 10:04:03 2024 (2046460666): sent bytes: 111 including checksum value: 18 
Sun Jun 30 10:04:03 2024 (2046460668): Requesting new panasonic extra data
Sun Jun 30 10:04:03 2024 (2046460671): Already sending data. Buffering this send request
Sun Jun 30 10:04:03 2024 (2046461119): Received 203 bytes data
Sun Jun 30 10:04:03 2024 (2046461122): Checksum and header received ok!
Sun Jun 30 10:04:03 2024 (2046461131): received TOP115 Water_Pressure: 2.54
Sun Jun 30 10:04:03 2024 (2046461134): Sending command from buffer
Sun Jun 30 10:04:03 2024 (2046461136): sent bytes: 111 including checksum value: 1 
Sun Jun 30 10:04:05 2024 (2046463263): Previous read data attempt failed due to timeout!
Sun Jun 30 10:04:05 2024 (2046463265): Received 0 bytes data
Sun Jun 30 10:04:06 2024 (2046463919): Heishamon stats: Uptime: 23 days 16 hours 27 minutes 43 seconds ## Free memory: 61% ## Heap fragmentation: 14% ## Max free block: 20504 bytes ## Free heap: 23608 bytes ## Wifi: 100% (RSSI: -46) ## Mqtt reconnects: 11946 ## Correct data: 97.06% Rules active: 0
Sun Jun 30 10:04:06 2024 (2046463939): Requesting new panasonic data
Sun Jun 30 10:04:06 2024 (2046463942): sent bytes: 111 including checksum value: 18 
Sun Jun 30 10:04:06 2024 (2046463949): Requesting new panasonic extra data
Sun Jun 30 10:04:07 2024 (2046464836): Already sending data. Buffering this send request
Sun Jun 30 10:04:07 2024 (2046464850): Received bad header. Ignoring this data!

[many identical rows deleted]

Sun Jun 30 10:04:07 2024 (2046465325): Received bad header. Ignoring this data!
Sun Jun 30 10:04:07 2024 (2046465334): Received 203 bytes data
Sun Jun 30 10:04:07 2024 (2046465335): Checksum and header received ok!
Sun Jun 30 10:04:07 2024 (2046465344): received TOP115 Water_Pressure: 2.56
Sun Jun 30 10:04:07 2024 (2046465362): Sending command from buffer
Sun Jun 30 10:04:07 2024 (2046465364): sent bytes: 111 including checksum value: 1 
Sun Jun 30 10:04:08 2024 (2046465823): Received 203 bytes data
Sun Jun 30 10:04:08 2024 (2046465824): Checksum and header received ok!
Sun Jun 30 10:04:11 2024 (2046468921): Heishamon stats: Uptime: 23 days 16 hours 27 minutes 48 seconds ## Free memory: 69% ## Heap fragmentation: 21% ## Max free block: 21152 bytes ## Free heap: 26760 bytes ## Wifi: 100% (RSSI: -45) ## Mqtt reconnects: 11946 ## Correct data: 97.06% Rules active: 0
Sun Jun 30 10:04:14 2024 (2046471488): Requesting new panasonic data
Sun Jun 30 10:04:14 2024 (2046471490): sent bytes: 111 including checksum value: 18 
Sun Jun 30 10:04:14 2024 (2046471492): Requesting new panasonic extra data
Sun Jun 30 10:04:14 2024 (2046471499): Already sending data. Buffering this send request
Sun Jun 30 10:04:14 2024 (2046471927): Received 203 bytes data
Sun Jun 30 10:04:14 2024 (2046471928): Checksum and header received ok!
Sun Jun 30 10:04:14 2024 (2046471937): Sending command from buffer
Sun Jun 30 10:04:14 2024 (2046471938): sent bytes: 111 including checksum value: 1 
Sun Jun 30 10:04:15 2024 (2046473021): Received 203 bytes data
Sun Jun 30 10:04:15 2024 (2046473022): Checksum and header received ok!

[Here the MQTT connection gets lost]

Sun Jun 30 10:04:16 2024 (2046473920): Lost MQTT connection!
Sun Jun 30 10:04:16 2024 (2046473921): Reconnecting to mqtt server ...
Sun Jun 30 10:04:19 2024 (2046477114): Heishamon stats: Uptime: 23 days 16 hours 27 minutes 57 seconds ## Free memory: 57% ## Heap fragmentation: 8% ## Max free block: 20504 bytes ## Free heap: 22216 bytes ## Wifi: 100% (RSSI: -46) ## Mqtt reconnects: 11947 ## Correct data: 97.06% Rules active: 0
Sun Jun 30 10:04:19 2024 (2046477136): Requesting new panasonic data
Sun Jun 30 10:04:19 2024 (2046477137): sent bytes: 111 including checksum value: 18 
Sun Jun 30 10:04:19 2024 (2046477144): Requesting new panasonic extra data
`

IgorYbema, 2024-06-30:
97% is fine, could be better but should not be any problem. And MQTT reconnects could be due to a busy wifi network but you could also ignore that as long it is not intefering with the information from and to heishamon.

klausj1, 2024-07-06:
What I do not understand is that the Heishamon says 97% good reads, wheres the HA extension says 747k good reads and 132k bad header reads. This is much worse than 97%. But OK, THats somehow fine.

Regarding the mqtt connection losses:

I have 4 other devices on my mqtt server, none of them complains. My Wifi network is also for sure not so busy so often.

Now, 6 days later, I have 14.574 mqtt reconnects, after 11.946. Thats 2.628 reconnects in 6 days, so 18 per hour, something is wrong here. What could that be? Is there any timeout setting which can be adjusted?

IgorYbema, 2024-07-06:
As a datastream contains 203 bytes, when the first byte (header) is wrong, it reads all other 202 bytes as wrong header also as it can not know if there is a real header somewhere in that datastream. That is why you see more bad header reads compared to the percentage good reads.

Heishamon has a very low mqtt timeout (5 seconds). So if there is no response within 5 seconds, heishamon disconnects. This is done to keep other funcitons running and not wait for mqtt to come back (as it is singled threaded). If you can try to disable the old 802.1b on your wifi. This greatly improves wifi wait times.

klausj1, 2024-07-07:
OK, something is wrong with the WiFi-Connection to the Heishamon in general, not only for MQTT.

Here is an example ping-result from my PC:
```
Antwort von 192.168.1.15: Bytes=32 Zeit=6ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=5ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.15: Bytes=32 Zeit=13ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=24ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.15: Bytes=32 Zeit=4ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=4ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=4ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.15: Bytes=32 Zeit=183ms TTL=255
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.1.15: Bytes=32 Zeit=95ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=246ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=6ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=3ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=6ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=4ms TTL=255
Antwort von 192.168.1.15: Bytes=32 Zeit=34ms TTL=255
```

I played around with the settings of my Fritzbox regarding which WiFi standards to support, did not change anything. I turned off the Fritz Repeater near the Heishamon - then the WIFI signal strength on the Heishamon went from 100% to around 60%, and it seemed there are more interruptions. Security is set to WPA2 (CCMP).

I have many other WiFi-Clients in my network (a Raspberry Pi, some ESP-32-based devices (Faikin, ESP32 DevKits)), none of them has this problem. But I do not have another ESP8266.

Any ideas?

klausj1, 2024-07-07:
I also tried to increase the settings "How often new values are collected from heatpump" from 5 seconds to 20 seconds and "How often all heatpump values are retransmitted to MQTT broker" from 300 to 600 seconds, but that did not help as well.

Also opening the door from the indoor heatpump unit (so that the heishamon is not inside the closed unit) did not change anything.

In the meantime (I do not know why), things are even worse, and the heatpump is more often in the state offline than online.

Its also pretty hard to look at the heishamon, because the webinterface often does not work because of the connection losses.

klausj1, 2024-07-31:
Any news on this? The connection losses happen more often than once per minute, if I want to control something via MQTT this will not work well ...

[image]

Coowly, 2024-08-08:
I have the same issue/ behavior with 1 thing extra; often the heishamon seems to stall and no data att all is coming through

IgorYbema, 2024-08-08:
If heishamon stalls it is also due to excessive wifi problems. It is really not something seen regular. Your problems should be looked into your own network. Did you enclose the heishamon within the heatpump maybe? This will weaken the wifi (even if it shows 100%).
It is hard to tell without looking physically at your location what is going on.

klausj1, 2024-08-11:
I did some other tests today:

- I left the door of the heatpump open so that the Heisamon is "free"
- When I put my fritzbox in the same room as the Heishamon, it works perfectly fine. Unfortunately this is no option, as I do not have a telephone cable (DSL) in this room and therefore no internet, this was just to test
- In the room with the Heishamon I have a fritzbox repeater 600. I tried with the repeater in the mesh, with the repeater not in the mesh, and without the repeater - all these options do not work, same problem as described above
- In the same room as the Heishamon I have a Raspberry Pi and an ESP32, both without problems

Any ideas? Maybe change to the eps32-based version?

IgorYbema, 2024-08-11:
Maybe heishamon doesn't connect to the repeater even if it is online. I am not familiar with the  fritzbox repeater 600, can it create a own wifi network itself also so you can configure heishamon to connect to the repeater?

klausj1, 2024-08-11:
The heishamon connects, and then disconnects, and then connects, ... It connects to the repeater, I can see this in the fritzbox UI.

The fritzbox 600 just extends the existing WLAN, it does not create an own access point.

IgorYbema, 2024-08-11:
Clearly it is not working correctly with the repeater. I am not sure why and I have never heard anyone about this before.  Maybe it helps to switch the channel?

 The ESP32 version has an ethernet option so you can skip wifi completly. That could be an option for you maybe.

klausj1, 2024-08-11:
Ich hab mal eine Supportanfrage an AVM (Fritzbox) geschickt, mal schauen ... und dann werde ich wohl die ESP32-Variante bestellen.

Coowly, 2024-08-12:
It seems not a wifi problem at my end.  The Heishamon is not enclosed in the heat pump and it is always reachable as also I see in the console that it keeps trying te get dat but no data is coming through. Only after a powersurge I get 1 or 2 times data and then right back to the seem probleem. After another power cyclus exactly the same. 
The wifi point is within 2 meters of the heishamon and is directly connected with a cable. I even set up so that the heishamon wil always connect to the closest point (I tried al the athletes also but no difference/improvement)
[image]

geduxas, 2024-08-12:
@Coowly which version do you run? Also which hardware? Do you have expansion board in heat pump?

Coowly, 2024-08-12:
[image]
[image]
Wifi- Asus ZenWifi XT8

Do you need more?

Coowly, 2024-08-12:
I have reset everything to install everything from the beginning. Unfortunately still the same. For a row moments I get data and then again the same, no data anymore. I really don't know what it can be or how to fix it. Everything has worked fine for almost 2 years now

IgorYbema, 2024-08-12:
@Coowly please confirm you have connected it to the correct port on the heatpump. Mainly, if you have the optional PCB installed (that one for multi zone support etc) you MUST connect heishamon to that board and not the main board.

Coowly, 2024-08-12:
Confirm. Everything is installed correct. It also shows exactly the same behavior with or without the optional PCB. I do get data, but after a few minutes it stops and I only get time outs. After a power cycle it works for a few moments, I get a few data updates (sometimes up to 20 minutes) before it goes back to no data/ time out

klausj1, 2024-08-22:
Just fyi: Updating the Fritzbox to a beta-version did not help. Then I did a factory reset of the Fritzbox, and restored the settings from backup, and now its working. No connection loss since two days. No idea why, I hope it stays this way. And definitely not a Heishamon issue, but a Fritzbox issue ... From my point of view this can be closed, bot @Coowly 's problem is still open probably.

klausj1, 2024-10-10:
Closing, as its a fritzbox / repeater issue

closed 2024-10-10 by klausj1