heishamon/HeishaMon #502: Updating firmware to 3.2.3 -> 3.5, FW update process in general

MEKadan, 2024-08-14:
I apologize, if this text goes to somewhat wrong place, but I could not find a better one, not least because of the fact that I'm quite unexperienced Github user and not a sw professional. Last time working with Heishamon installation was some time ago, and I cannot remember what was done then.

My question is about installing new firmware. The Readme does not contain very detailed instructions on updating FW

First of all, I got the following message in Home Assistant:

[image]

After exploring a while, I ended up in https://github.com/Egyras/HeishaMon/tree/master/binaries

If I am not mistaken, I should pick up a suitable binary file, which I suspect is the HeishaMon.ino.d1-v3.5.bin and install it via Heishamon WebUI.  

[image]

The file (3.5) is in the mode-type-small folder, however, I cannot find the corresponding md5? 

[image]

The Readme in binaries -folder talks about "The LittleFS versions" Is this the same than mode-type-small?

Then, the Readme syas: "The LittleFS versions will, after updating to this version, reset your HeishaMon to factory default due to change to a new filesystem for the configuration."

Is it enough to restore my original configuration just by copying all the parameters from the Settings -page of the WebUI?

geduxas, 2024-08-14:
I think it was forgotten to merge.. @IgorYbema ?

Could you try this md5: https://github.com/IgorYbema/HeishaMon/blob/main/binaries/mode-type-small/HeishaMon.ino.d1-v3.5.md5

MEKadan, 2024-08-15:
OK, thanks, @geduxas 

The update seem to reset pcb to factory settings, so is it enough to restore my original configuration just by copying all the parameters from the Settings -page of the WebUI?

IgorYbema, 2024-08-15:
It should not reset it to factroy settings but when it does you can easily restore the settings indeed by just placing them again in the settings page.

MEKadan, 2024-08-16:
> I think it was forgotten to merge.. @IgorYbema ?
> 
> Could you try this md5: https://github.com/IgorYbema/HeishaMon/blob/main/binaries/mode-type-small/HeishaMon.ino.d1-v3.5.md5
> 

I tried this, and it informed the update failed because of md5

usernr1, 2024-08-16:
Had the same issue and skipped v3.5. 
v3.6 works for me without problems.
[https://github.com/IgorYbema/HeishaMon/blob/main/binaries/mode-type-small/HeishaMon.ino.d1-v3.6.bin](url)
MD5 v3.6`1a61f27340b6b07f8f92562f6382067a`
[image]

MEKadan, 2024-08-16:
So You guys are fetching these from IgorYbema's repo, not from here?

jvuotila, 2024-08-17:
I did update and now I cannot connect to Heishamon at all. I have tried to push reset button on the device and can not find even initial hotspot. What can I do now?

IgorYbema, 2024-08-17:
> I did update and now I cannot connect to Heishamon at all. I have tried to push reset button on the device and can not find even initial hotspot. What can I do now?

Did you use the correct file for your heishamon model (large or small version)?

jvuotila, 2024-08-17:
> > I did update and now I cannot connect to Heishamon at all. I have tried to push reset button on the device and can not find even initial hotspot. What can I do now?
> 
> Did you use the correct file for your heishamon model (large or small version)?

I used large version (HeishaMon.ino.esp32-v3.5.bin). But any tips how to solve problem?

IgorYbema, 2024-08-17:
You used the large version on the small module I assume? (as the large version was already installed with v3.5 from shipping).

To recover the small version you need a USB-TTL cable and you plug that on the correct pins on the 10-pin header (gnd, rx,tx and 5v).  When connected you put the heishamon in firmware update mode by holding the boot button while resetting it. You can release the boot button after the reset. It should be in firmware update mode now. Then use https://espressif.github.io/esptool-js/ to upload the correct file
[image]

jvuotila, 2024-08-17:
Thanks, I made an order for the cable. Thank god it's not winter and heating season. 😆

MEKadan, 2024-08-17:
Hmmm... based on how Your cable looks, there are 3,3V regulators onboard? Probably would go even without, ESP's seem to tolerate TTL level well, e.g. I have been running 2 separate ESP's contacted directly to TTL level serial lines for 2 years day and night without problems. 

I once flashed Heishamon via cable with 3,3V converter chip in between, just because I happen to have one. Goes OK as well

IgorYbema, 2024-08-17:
Just for firmware flashing the 5v ttl is ok

TherealGunnar, 2024-08-18:
> I used large version (HeishaMon.ino.esp32-v3.5.bin). But any tips how to solve problem?

I was stupid enough to do the same. Also bought the cable. I agree that a good explanation on how to update is missing.

IgorYbema, 2024-08-18:
I updated the readme. Is this clear enough now? https://github.com/IgorYbema/HeishaMon/tree/main/binaries

MEKadan, 2024-08-18:
On the main page https://github.com/IgorYbema/HeishaMon/tree/main still reads: 

`Current release is version 3.5. `

On the right, in the releases section we can see:

[image]

No release information later releases

Then, currently Home Assistant shows an available update, leading through link "Read release announcement" currently to here: https://github.com/Egyras/HeishaMon/releases/tag/v3.5

[image]

I'm still wondering, what actually is the role of each of these 2 repos (Egyras/ IgorYbema)?

Where should we primarily be downloading updates from?

IgorYbema, 2024-08-18:
If you want to follow stable release the Egyras repo is the way to go. When you follow my repo you will get more recent updates (no releases). But eventually I am the only maintainer. Has been like this for years now.

MEKadan, 2024-08-18:
Ahh... OK. Thanks. Nice work!

Pusch3n, 2024-08-19:
> You used the large version on the small module I assume? (as the large version was already installed with v3.5 from shipping).
>
> To recover the small version you need a USB-TTL cable and you plug [...]

Hey,

i made the same stupid mistake 🙈  I tried to fix it with the USB-TTL cable. The programming also finished without an error. The Problem is I dont find the Wlan AP of heishamon after the flashing. I tried 3.2.3 and 3.5. 
Do you have a idea what went wrong?

Thanks

IgorYbema, 2024-08-19:
> i made the same stupid mistake 🙈 I tried to fix it with the USB-TTL cable. The programming also finished without an error. The Problem is I dont find the Wlan AP of heishamon after the flashing. I tried 3.2.3 and 3.5. Do you have a idea what went wrong?

Are you sure you flashed the correct binary this time? What activity is visible on the blue led?

Pusch3n, 2024-08-19:
This one: [HeishaMon.ino.d1-v3.2.3.bin](https://github.com/Egyras/HeishaMon/blob/master/binaries/mode-type-small/HeishaMon.ino.d1-v3.2.3.bin)

also the 3.5 bin from the same folder.

The blue led is off, except when I press the reset button. 
How should it behave?

IgorYbema, 2024-08-19:
Should work. Are you flashing to address 0x0 ? Not sure what the website flash tool selects as default address. 
Blue led should start blinking. Did you try factory reset already?

Pusch3n, 2024-08-19:
On the website the default is 0x1000. I will try 0x0.

Thanks alot

jvuotila, 2024-08-20:
I also trying to flash correct firmware and only got this error message on ESP Tool terminal. Any hints? Tried to flash 0x0 and 0x1000 address.

Error: Cannot read properties of undefined (reading 'BOOTLOADER_FLASH_OFFSET')

IgorYbema, 2024-08-20:
Aren't you getting this?
[image]

jvuotila, 2024-08-20:
Sorry, deleted my comment... My own mistake. Rx and tx pins were incorrect position.

jvuotila, 2024-08-20:
Everything is working again. Thanks to everyone on this feed! 🙌🏻

jllussa, 2024-08-21:
Very useful thread @IgorYbema, consider it some cut and paste to the main README on https://github.com/Egyras/HeishaMon
and adding section... "How to unbrick Heishamon"  
Program used -> https://espressif.github.io/esptool-js/  
Starting at -> HEX 00

Must be in!  :)

TherealGunnar, 2024-08-22:
I get the error Error: CryptoJS is not defined while trying to flash this one: [HeishaMon.ino.d1-v3.2.3.bin](https://github.com/Egyras/HeishaMon/blob/master/binaries/mode-type-small/HeishaMon.ino.d1-v3.2.3.bin)

Stueckgold, 2024-09-25:
I try to flash from 3.2.3 to 3.8 or any other binary:
I always get errors 
"MD5 wrong"
or without md5 it´s
"Update failed! Please try again...ERROR[10]: Magic byte is wrong, not 0xE9"

small version

closed 2024-10-31 by github-actions[bot]