heishamon/HeishaMon #503: reboot with my rules set since HeishaMon.ino.bin #544 Alpha-e17aac1

blb4github, 2024-08-19:
Hi,

I'm using already longer time alpha releases together with my rules set. The last version that functions properly is version #544 Alpha-e17aac1. Since that version I have tested several build but all are rebooting direct after rules are loaded and then boot without rules.

debuglog:

Settings conditionals...
Mon Aug 19 21:39:42 2024 (1212): Number of 1wire sensors on bus: 5
Mon Aug 19 21:39:44 2024 (3295): Found 1wire sensor: 2844f50d12210100
Mon Aug 19 21:39:44 2024 (3295): Found 1wire sensor: 282c5415122101b8
Mon Aug 19 21:39:44 2024 (3296): Found 1wire sensor: 288efc9312210157
Mon Aug 19 21:39:44 2024 (3296): Found 1wire sensor: 2889658f122101bc
Mon Aug 19 21:39:44 2024 (3296): Found 1wire sensor: 283f13071221013d
Settings DNS...
Check OT config...
E (2978) gpio: gpio_install_isr_service(500): GPIO isr service already installed
Enabling rules..
Reset reason: 3
4315: rules mempool size: 32768
4315: reading rules
4388: rule #1 was prepared in 59764 microseconds
4554: rule #1 bytecode was created in 165195 microseconds
4554: bytecode: 168/168, heap: 32/32, stack: 4/4 bytes, varstack: 272/272 bytes
4555: timer #1 set to 60 seconds
4555: timer #2 set to 10 seconds
4555: rule #1 was executed in 841 microseconds
4555: bytecode: 168/168, heap: 32/32, stack: 4/12 bytes, varstack: 272/272 bytes
4923: rule #2 was prepared in 368286 microseconds
5587: rule #2 bytecode was created in 663399 microseconds
5587: bytecode: 780/780, heap: 116/116, stack: 4/12 bytes, varstack: 400/400 bytes
5590: timer #8 set to 60 seconds
5590: TaManipulations did run
5590: rule #2 was executed in 2398 microseconds
5590: bytecode: 780/780, heap: 116/116, stack: 4/16 bytes, varstack: 400/400 bytes
5591: rule #3 was prepared in 497 microseconds
5592: rule #3 bytecode was created in 1143 microseconds
5592: bytecode: 40/40, heap: 20/20, stack: 4/16 bytes, varstack: 424/424 bytes
5593: rule #3 was executed in 378 microseconds
5593: bytecode: 40/40, heap: 20/20, stack: 4/16 bytes, varstack: 424/424 bytes
5593: rule #4 was prepared in 317 microseconds
5594: rule #4 bytecode was created in 793 microseconds
5594: bytecode: 28/28, heap: 12/12, stack: 4/16 bytes, varstack: 448/448 bytes
5595: rule #4 was executed in 283 microseconds
5595: bytecode: 28/28, heap: 12/12, stack: 4/16 bytes, varstack: 448/448 bytes
5631: rule #5 was prepared in 35460 microseconds
5672: rule #5 bytecode was created in 41208 microseconds
5672: bytecode: 312/312, heap: 88/88, stack: 4/16 bytes, varstack: 464/464 bytes
5674: timer #7 set to 600 seconds
5674: timer #7 set to 25 seconds
5674: OpenThermThermostat did run
5674: rule #5 was executed in 1446 microseconds
5674: bytecode: 312/312, heap: 88/88, stack: 4/16 bytes, varstack: 464/464 bytes
5723: rule #6 was prepared in 48987 microseconds
5833: rule #6 bytecode was created in 109531 microseconds
5833: bytecode: 388/388, heap: 72/72, stack: 4/16 bytes, varstack: 528/528 bytes
5836: timer #6 set to 900 seconds
5836: rule #6 was executed in 2855 microseconds
5837: bytecode: 388/388, heap: 72/72, stack: 4/16 bytes, varstack: 528/528 bytes
5988: rule #7 was prepared in 150982 microseconds
6180: rule #7 bytecode was created in 191899 microseconds
6180: bytecode: 524/524, heap: 108/108, stack: 4/16 bytes, varstack: 608/608 bytes
6183: timer #5 set to 60 seconds
6184: rule #7 was executed in 3352 microseconds
6184: bytecode: 524/524, heap: 108/108, stack: 4/20 bytes, varstack: 608/608 bytes
6184: rule #8 was prepared in 292 microseconds
6185: rule #8 bytecode was created in 797 microseconds
6186: bytecode: 28/28, heap: 12/12, stack: 4/20 bytes, varstack: 624/624 bytes
6186: rule #8 was executed in 237 microseconds
6186: bytecode: 28/28, heap: 12/12, stack: 4/20 bytes, varstack: 624/624 bytes
6248: rule #9 was prepared in 61968 microseconds
6335: rule #9 bytecode was created in 86362 microseconds
6335: bytecode: 428/428, heap: 88/88, stack: 4/20 bytes, varstack: 632/632 bytes
6336: timer #3 set to 120 seconds
6336: rule #9 was executed in 1022 microseconds
6336: bytecode: 428/428, heap: 88/88, stack: 4/20 bytes, varstack: 632/632 bytes
6421: rule #10 was prepared in 84310 microseconds
6596: rule #10 bytecode was created in 174309 microseconds
6596: bytecode: 292/292, heap: 56/56, stack: 4/20 bytes, varstack: 800/800 bytes
6597: rule #10 was executed in 1435 microseconds
6597: bytecode: 292/292, heap: 56/56, stack: 4/20 bytes, varstack: 800/800 bytes
6604: rule #11 was prepared in 6387 microseconds
6621: rule #11 bytecode was created in 16730 microseconds
6621: bytecode: 172/172, heap: 24/24, stack: 4/20 bytes, varstack: 904/904 bytes
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556b4  PS      : 0x00060a30  A0      : 0x8201c5e9  A1      : 0x3fceba50  
A2      : 0x00080119  A3      : 0x00080115  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0xffffc000  A9      : 0xffffffff  
A10     : 0x3fcebc90  A11     : 0x00000000  A12     : 0x00000008  A13     : 0x0000001d  
A14     : 0x00000019  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00080119  LBEG    : 0x400554b9  LEND    : 0x400554dd  LCOUNT  : 0x820111b3  

Backtrace: 0x400556b1:0x3fceba50 0x4201c5e6:0x3fceba60 0x42011229:0x3fceba90 0x4201d213:0x3fcebcf0 0x4201eb10:0x3fcebdc0 0x42011d46:0x3fcebe90 0x42006b6e:0x3fcebf70 0x4202f25f:0x3fcebfd0 0x40381266:0x3fcebff0

ELF file SHA256: 18b660c3188b8e2b

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037c026
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
Starting debugging, version: 3.6

--- HEISHAMON ---
starting...
Starting littlefs...
Started littlefs...
Checking littlefs for first boot...
Heishamon boot file exists, normal boot...
Send current wifi info to serial...
Mode: NULL
Channel: 0
SSID (0): 
Passphrase (0): 
BSSID set: 239
Loading config from flash...
Mon Aug 19 21:39:49 2024 (654): mounting FS...
Mon Aug 19 21:39:49 2024 (654): mounted file system
Mon Aug 19 21:39:49 2024 (661): reading config file
Mon Aug 19 21:39:49 2024 (668): opened config file
Mon Aug 19 21:39:49 2024 (673): {"wifi_hostname":"HeishaMon","wifi_password":"","wifi_ssid":"","ota_password":"heisha","mqtt_topic_base":"panasonic_heat_pump","mqtt_topic_listen":"master_panasonic_heat_pump","mqtt_server":"172.31.0.5","mqtt_port":"1883","mqtt_username":"","mqtt_password":"","use_1wire":"enabled","use_s0":"enabled","hotspot":"enabled","listenonly":"disabled","listenmqtt":"disabled","logMqtt":"disabled","logHexdump":"disabled","logSerial1":"enabled","optionalPCB":"disabled","opentherm":"enabled","proxy":"enabled","waitTime":5,"waitDallasTime":5,"dallasResolution":11,"updateAllTime":300,"updataAllDallasTime":300}
Mon Aug 19 21:39:49 2024 (676): parsed json
Setup wifi...
Mon Aug 19 21:39:49 2024 (677): Wifi reconnecting with new configuration...
Mon Aug 19 21:39:49 2024 (857): Wifi hotspot mode...
Setup ethernet module...
Setup MQTT...
Setup HTTP...
Sync webserver server started at port: 80
Setup SNTP...
Switch serial...
Sending new wifi diag...
Mode: AP

Binifada1956, 2024-08-19:
Has you also test V 3.6?

blb4github, 2024-08-19:
Yes, I tried until V3.6. I see V3.7 is released yesterday, I'll try that now

blb4github, 2024-08-19:
I've tried the latest builds as well (#567, #571), same result.

IgorYbema, 2024-08-20:
Please share your ruleset

IgorYbema, 2024-08-20:
Also please understand that #544 isn't in the latest releases yet. The PR for those changes https://github.com/IgorYbema/HeishaMon/pull/133 is still in testing so if you use advanced rules from that PR it will not work in the latest releases.

blb4github, 2024-08-20:
My rules set:
[Heishamon_rules20240819.txt](https://github.com/user-attachments/files/16673392/Heishamon_rules20240819.txt)

IgorYbema, 2024-08-20:
I believe those rules are indeed only compatible with the changes in the PR and therefore not yet working in the latest release. Please note your crashes in the PR

blb4github, 2024-08-20:
I have tested your latest version (https://github.com/IgorYbema/HeishaMon/actions/runs/10476556300) and I took out the print commands out of my rules set and now it works! I guess this one can be closed now.....
rules set that works now: Heishamon_rules20240819_without_print

IgorYbema, 2024-08-20:
Thanks for the notice

blb4github, 2024-08-20:
I close this one, it seems we can continue https://github.com/Egyras/HeishaMon/issues/486#issuecomment-2299824222

closed 2024-08-20 by blb4github

blb4github, 2024-08-20:
This can now continue https://github.com/IgorYbema/HeishaMon/pull/133#issuecomment-2292212997, Open point seems now the long rule as mentioned there