heishamon/HeishaMon #508: Problem with 1-wire at Heishamon V5 (FW 3.5)

WP-Rue, 2024-09-03:
After updating from Wifi-Heishamon to large Version V5 I mixed up 5V+ and 1-wire-signal when connecting 1-wire network (with 3 sensors).
Now I fixed it - but I get no signal from Heishamon: no sensor is found. Even a single new sensor didn't work. Reboot has no effect. 
Any other ideas? Is it possible, that something in the hardware is destroyed? 
Thank you in advanced...

IgorYbema, 2024-09-04:
I am not sure if that can destroy heishamon but if it isn't reading the sensor anymore it problably is. I do know that wiring it incorrectly can destroy the sensor itself (it gets very hot and then burns down) but you said you used a new sensor also.

WP-Rue, 2024-09-04:
Thanks for your answer.. I will try the old Wifi-Version again with a single sensor and try to compare the signals at the output with the V5-eth-Version to check, if I have a problem with the sensors or the HW...
Luckily I don't need the sensors essentially at the moment.

WP-Rue, 2024-09-04:
Good news- only 1 1-wire sensor defect: 2 Ohm between GND and the 1-wire terminal.
Everything works again!
It seems as if you need to reboot Heishamon to detect new sensors? 
(Or maybe I didn't wait long enough 😉)

closed 2024-09-04 by WP-Rue