heishamon/HeishaMon #510: UI stuck in "...Loading..." since 3.7

bagbag, 2024-09-06:
After updating to 3.7, the UI doesn't show any values anymore. The table is stuck in `...Loading...` state.

Network monitor of developer console says `SyntaxError: JSON.parse: bad control character in string literal at line 1 column 7300 of the JSON data` for the new json request.

Visual studio code tells me the following for the response:
[image]

Hex view:
[image]

bagbag, 2024-09-06:
The problem seems to be the tab after IDU: instead of a regular space. It has to be either a space or escaped to `\t`.

IgorYbema, 2024-09-06:
Thanks. I'll upload a new v3.7 asap (will keep it at v3.7, just look at different timestamp)

bagbag, 2024-09-06:
Great!
New binary from your repository works fine.

closed 2024-09-06 by bagbag