heishamon/HeishaMon #512: update WebSockets to 2.4.1 or later?

mi-hol, 2024-09-06:
platformio.ini currently uses 

links2004/WebSockets@^2.4.0

but release [2.4.1](https://github.com/Links2004/arduinoWebSockets/releases/tag/2.4.1) includes fixes like:
"Avoid buffer overflow for small Basic Authentication header by @matth-x in https://github.com/Links2004/arduinoWebSockets/pull/841
**Fix memory leak and improve performance** by @BetterComputing in https://github.com/Links2004/arduinoWebSockets/pull/852"

Therefore I wonder if the dependency should be at least be updated to this version or even the latest 2.6.0 for newer hardware support?

geduxas, 2024-09-06:
Main key word, MIGHT, and IN BASIC AUTH, which is not used in heishamon, yes it's better to use lastest lib's, but before it should run for testing. Heishamon is highly demand in resources, so everything could break it

IgorYbema, 2024-09-06:
You did point me at something :-)
We don't use the websockets library anymore. It has it own implementation (also due to memory requirements). But there is still some old code pointing to this so I need to remove that.

mi-hol, 2024-09-07:
closed via https://github.com/Egyras/HeishaMon/commit/00e5d4d42123ae9e03009f48d078db126433e86a

closed 2024-09-07 by mi-hol