heishamon/HeishaMon #513: Release version 3.7 contains "Alpha" + commit identifier, instead of actual release version

mi-hol, 2024-09-06:
https://github.com/Egyras/HeishaMon/blob/master/.github/workflows/main.yml sets 

```

 - name: Update version.h
        if: github.ref != 'refs/heads/main' #do not change version in main branch run
        run: cd HeishaMon && echo "static const char *heishamon_version = \"Alpha-$(git rev-parse --short HEAD)\";" > version.h && cat version.h
        shell: bash'''

```

Suggest a separate yaml for release build

geduxas, 2024-09-06:
Nope release contains correct build 
[image]

IgorYbema, 2024-09-06:
Only if you take the build from main or directly the file from the binaries folder you will see proper v3.7

mi-hol, 2024-09-06:
in the meantime I downloaded and opened for example https://github.com/Egyras/HeishaMon/blob/master/binaries/model-type-large/HeishaMon.ino.esp32-v3.7.bin in an editor

There is indeed a proper version number

[image]

closed 2024-09-06 by mi-hol