heishamon/HeishaMon #515: Publish optional PCB values to mqtt

bagbag, 2024-09-06:
We can set parameters for optional PCB like `DemandControl` and `SmartGridMode`.
Is there any reason for them to be not published via mqtt?

This would be useful for example for home assistant to display the value which heishamon is currently sending to the heatpump (https://github.com/kamaradclimber/heishamon-homeassistant/pull/238).

bagbag, 2024-09-06:
But it's only controlled by heishamon, isn't it?
So heishamon should always know the latest value - except if heishamon reboots.

But even if heishamon doesn't know it, the lastest published mqtt message is still there, which should reflect the real value - except if the heat pump reboots. But that are edge cases, which shouldn't matter and hold this feature back entirely.

Am I missing something?

geduxas, 2024-09-06:
Yes, and no.. heishamon works in extension PCB emulator.. so it's trying to mimic real PCB behavior.. in real PCB you have 0-10V input for demand control, and according to voltage, it's sends to pump set value... So status report could theoretically work just in PCB emulation mode. Also, you can keep same value you have send to heishamon itself.. so don't see any point to report it back.

mi-hol, 2024-09-06:
> keep same value you have send to heishamon itself.. so don't see any point to report it back.

is along the lines of yet unpublished design decision as explained in https://github.com/Egyras/HeishaMon/issues/507#issuecomment-2323209215

bagbag, 2024-09-06:
> heishamon is a gateway and doesn't store any state information

is a good point, so this is out of scope.

closed 2024-09-06 by bagbag