heishamon/HeishaMon #518: Static IP

krzbor, 2024-09-07:
For devices that I can connect to (e.g., via HTTP), I use static IP addresses outside of the DHCP pool. Is it possible to add static IP configuration (including mask and gateway, of course) in HeishaMon?

IgorYbema, 2024-09-07:
Yes I'll add that to my list of todo's :)

krzbor, 2024-09-07:
Is the list long? :)

stumbaumr, 2024-09-09:
When using DHCP just assign a fixed IP to the MAC address in your DHCP server.
Best practise: Maintaining IPs centrally is much easier in the long run than setting them on the devices directly.

krzbor, 2024-09-10:
I've changed routers many times in my life. The manufacturers of this equipment have also changed several times. However, I have NEVER changed the addressing of my network. The answer is simple - a static IP is a better solution than DHCP. The pool of private addresses was defined in the last century and hasn't really changed. So what is more permanent - the router with its configuration or the static IP address defined in RFC1918?

geduxas, 2024-09-10:
Yes with static IP you could survive while recovering dhcp router ;)

stumbaumr, 2024-09-10:
You change the router from Speedport to Fritzbox, you use the old WLAN and PSK in the new router, but then you realize the IP address range is different...

When your DHCP router reboots/recovers your other devices do not suddenly loose the currently assigned IPs...

geduxas, 2024-09-10:
@stumbaumr is it really? Manufactures blocks ip range? :D i thought it's fing form far far past :D

krzbor, 2024-09-10:
For my devices – the server, virtual machines, an Ethernet RS485 controller, cameras, underfloor heating control, IP intercom, and additional APs – I have static IP addresses, which I document in a file. They remain unchanged, even when I replace the router. Now it's probably clear why I want to add Heishamon to this "collection".

stumbaumr, 2024-09-10:
> For my devices – the server, virtual machines, an Ethernet RS485 controller, cameras, underfloor heating control, IP intercom, and additional APs – I have static IP addresses, which I document in a [...]

We are fighting in IgorYbema's repo against memory issues causing reboots and here you request features no one really needs. You should print that file out on paper and laminate it...

krzbor, 2024-09-10:
Now I understand – you're criticizing all my ideas just because you want IgorYbema to focus on an issue that's important to you – memory problems and reboots. From experience, I know that tracking down such issues is a complex task. In the case of using MQTT, reboots don't matter much since the ESP restarts in just a few seconds. DeepSleep for this processor is a commonly used feature in battery-powered solutions, and every return from DeepSleep is nothing more than a reboot. If you think no one needs a static IP, then why is the option to configure a static IP available in almost all network-controlled devices (e.g., those with a built-in web server)?

geduxas, 2024-09-10:
It's not Igor repo, there is no need to fight, static ip should be implemented, don't think there is any problem with that. It just needs time. 
Why should anyone prove why one or another function is important for them. I didn't see anywhere system which works only on DHCP..

stumbaumr, 2024-09-10:
As always with opensource... just clone, build and ask for the pull request to be merged...

krzbor, 2024-10-16:
Do you happen to know when this feature will be available?

closed 2024-11-26 by github-actions[bot]