heishamon/HeishaMon #519: Add design decision to documentation

mi-hol, 2024-09-08:
I'm very new to HeishaMon but have already seen a number of issues that seem caused from a lack of understanding HeishaMon's architecture, design goals and limitations.

**Proposal:**
Therefore I'd suggest to add  HeishaMon's architecture, design goals and limitations to main ReadMe.

**Benefits:**
- better educated users
- better decisions for users regarding "fit for purpose"
- knowledge sharing for good practice regarding implementation of automation
- fewer issues to deal with
- less conflicts with users & contributors

**Examples:**

- https://github.com/Egyras/HeishaMon/issues/507#issuecomment-2323036811 

- https://github.com/Egyras/HeishaMon/issues/515#issuecomment-2334312485

geduxas, 2024-09-08:
As gateway is mostly finished product..  almost all telemetry is well decoded and published.. so almost All issues could be within small fixes, or unimportant.. as for model naming in telemetry.. truly Panasonic don't have exect model mame or numeber in telemetry.. it have some kind of firmwar/configuration checksums which mostly corespondents to pump type.. but as much we knowt about it.. more we see same pump's with bit different ID's..

mi-hol, 2024-09-08:
@geduxas thanks for the comment, unfortunately I don't get what you are trying to tell me.
I'm a simple guy, preferring simple yes/no answers :)

geduxas, 2024-09-08:
> @geduxas thanks for the comment, unfortunately I don't get what you are trying to tell me. I'm a simple guy, preferring simple yes/no answers :)

Sorry :) your welcome to know whis project better, and understand everything soon i think. Yes it's missing some documentation, and yes it requires a little bit advanced experience.. simply it's HeatPump<->MQTT gateway.. so everything what pump spits to interface is translated and published into MQTT.. 

Maybe you could contribute this project and help with documentation? From andvanced user's, it's seam's everything simple and we'll known.. but yes for newcomers it's bit confusing. For my self, i have HeatPump for over 3y, also tried to reverse engineering protocol, and native a2w gateway.

mi-hol, 2024-09-08:
> Maybe you could contribute this project and help with documentation?

Happy to do so and made already some small contributions

mi-hol, 2024-09-11:
@Egyras @IgorYbema may I kindly ask for your view on my proposal?
Wanted to avoid investing time into a topic that is not seen as adding value by the main drivers of project

closed 2024-10-22 by github-actions[bot]