heishamon/HeishaMon #522: Creation of a topic for 1-wire thermometers using an alias

krzbor, 2024-09-08:
A very interesting feature of Heishamon is reading the temperature via 1-wire and sharing this value through MQTT. A special topic is created, for example: panasonic_heat_pump/1wire/2861755fb1220712. This topic can be used by various types of automation. However, the problem arises when replacing a faulty sensor with another one. In such a case, a new topic is generated, and the entire automation needs to be adjusted. Heishamon has a mechanism for assigning aliases to 1-wire sensors. Is it possible to add a new topic in the format: panasonic_heat_pump/1wire_alias/water_temperature, where "water_temperature" is the alias for the 1-wire sensor? This type of topic would only be available if the alias has been assigned, is unique, and complies with MQTT rules.

geduxas, 2024-09-08:
I am also vote for this feature, and also would like to extend with calibration possibility.

IgorYbema, 2024-09-08:
That i indeed the reason why I added the alias option. Currently it is added as a subtopic of the dallas-id in mqtt but that doesn't make it very usefull yet.
For backwards compatibility I'll keep adding the current method but I'll add the alias mqtt topic you described.

krzbor, 2024-09-08:
I'm very glad that the idea was well received. I wrote that the alias should comply with MQTT rules. I meant special characters. I'm wondering if it might be worth making one exception – for the "/" character. This would make it easy to create a subtopic.

jllussa, 2024-09-11:
I'm also using 1-wire sensor inside rules... so an alias would be great if ever needs to be "hardware" replaced.

on timer=2 then   
  $SensorBuffer = round(ds18b20#28ff641e8537c1eb);

krzbor, 2024-10-16:
Do you happen to know when this feature will be available?

krzbor, 2024-11-25:
I have prepared the appropriate change in the program:
https://github.com/Egyras/HeishaMon/pull/590