heishamon/HeishaMon #523: Switching the mode from the heating/cooling curve to direct value setting and vice versa.

krzbor, 2024-09-08:
Heishamon has the ability to check the operating mode: Heating_Mode and Cooling_Mode. However, I couldn't find information on how to switch the mode from the heating/cooling curve to direct value setting and vice versa. Is this possible?

IgorYbema, 2024-09-09:
There is no command for that yet. Do you think it is usefull to have?

MiG-41, 2024-09-09:
True , it is technically possible ,but this command wouldn't be used frequently ,especially in automation , you set it once on HP and don't touch it later ever... Or you have other ideas ?

krzbor, 2024-09-09:
If we are talking purely about automation, such switching doesn't really make sense. However, Heishamon can be used for remote control of a heat pump, for example, when we are not at home or when we are helping others for whom it is all too complicated (e.g., older people). It may become necessary when we can't achieve the correct temperature using the (-5,+5) adjustment, and we don't want to change the heating curve every time. I will be using the heat pump for cooling (chilled water and water cooler). Here, the amount of cooling depends on the water temperature but also on the airflow. This requires testing to see whether a cooling curve or a constant temperature works better. In general, Heishamon allows setting both the heating curve and the constant temperature. However, you cannot switch between these modes, which seems like a bit of a flaw.

stumbaumr, 2024-09-09:
Using the heating curve you can emulate direct value setting. But when using direct value setting you can not use the heating curve or use the heating curve shift (which is very useful when trying to prolong compressor runtimes).

Best practise: When using HeishaMon it is always best to use a heating/cooling curve!

krzbor, 2024-09-09:
"Using the heating curve you can emulate direct value setting." You're right, but why emulate something through the heating curve when the heat pump allows switching to a constant temperature? Wouldn't it be simpler to just switch from one mode to the other?

stumbaumr, 2024-09-09:
If it is so simple: Fork it, build it and create a pull request.

McMagellan, 2024-09-10:
If I saw it correctly, when changing the operating mode from heating curve to fixed value or vice versa, an area in the HP control is normalized and thus deleted. It affects the heating curve parameters and the entire weekly timer is completely deleted. There the value ranges must be changed from absolute to relative and existing data is deleted or set to factory settings. This makes the benefit of a change questionable and whether this process can be changed during operation at all without disrupting ongoing operations would have to be tested.

krzbor, 2024-09-10:
I didn't know about that. I can't check it now because I've started the process of drying the concrete floor. However, if it's as you say, then indeed switching from the heating curve to a fixed value and vice versa makes less sense. Maybe someone can confirm how it actually works?

geduxas, 2024-09-10:
heating curve parameters should still be in configuration. Because they are independent, and accessible by telemetry independent heating mode.. (never tested by myself) 

as for time scheduler, it's really interesting how it would work.. because scheduler logic are stored in remote controller..

krzbor, 2024-10-16:
I have decided to switch to heating/cooling curves. Although I now have to set 4 values, I can always set a flat curve (constant temperature), so it's a universal solution.

closed 2024-11-25 by github-actions[bot]