heishamon/HeishaMon #524: ESP8266 Compilation error: 'ArduinoJson6194_F1::JsonDocument::JsonDocument()' is protected within this context

Maystero14, 2024-09-09:
I want compilogin firmware for the board ESP8266 D1. During compiling I have mesage:
`C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp: In function 'void getWifiScanResults(int)':
C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:59:18: error: 'ArduinoJson6194_F1::JsonDocument::JsonDocument()' is protected within this context
   59 |     JsonDocument wifiJsonDoc;
      |                  ^~~~~~~~~~~
In file included from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:28,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.h:13,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:1:
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:308:3: note: declared protected here
  308 |   JsonDocument() : _pool(0, 0) {
      |   ^~~~~~~~~~~~
C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:59:18: error: 'ArduinoJson6194_F1::JsonDocument::~JsonDocument()' is protected within this context
   59 |     JsonDocument wifiJsonDoc;
      |                  ^~~~~~~~~~~
In file included from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:28,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.h:13,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:1:
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:320:3: note: declared protected here
  320 |   ~JsonDocument() {}
      |   ^
C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:65:65: error: no matching function for call to 'ArduinoJson6194_F1::ArrayRef::add<ArduinoJson::JsonObject>()'
   65 |       JsonObject wifiJsonObject = wifiJsonArray.add<JsonObject>();
      |                                                                 ^
In file included from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantShortcuts.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRef.hpp:18,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:24,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.h:13,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:1:
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayShortcuts.hpp:33:21: note: candidate: 'bool ArduinoJson6194_F1::ArrayShortcuts<TArray>::add(const T&) const [with T = ArduinoJson6194_F1::ObjectRef; TArray = ArduinoJson6194_F1::ArrayRef]'
   33 |   FORCE_INLINE bool add(const T &value) const {
      |                     ^~~
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayShortcuts.hpp:33:21: note:   candidate expects 1 argument, 0 provided
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayShortcuts.hpp:40:21: note: candidate: 'bool ArduinoJson6194_F1::ArrayShortcuts<TArray>::add(T*) const [with T = ArduinoJson6194_F1::ObjectRef; TArray = ArduinoJson6194_F1::ArrayRef]'
   40 |   FORCE_INLINE bool add(T *value) const {
      |                     ^~~
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayShortcuts.hpp:40:21: note:   candidate expects 1 argument, 0 provided
C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp: In function 'void loadSettings(settingsStruct*)':
C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:200:22: error: 'ArduinoJson6194_F1::JsonDocument::JsonDocument()' is protected within this context
  200 |         JsonDocument jsonDoc;
      |                      ^~~~~~~
In file included from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:28,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.h:13,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:1:
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:308:3: note: declared protected here
  308 |   JsonDocument() : _pool(0, 0) {
      |   ^~~~~~~~~~~~
C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:200:22: error: 'ArduinoJson6194_F1::JsonDocument::~JsonDocument()' is protected within this context
  200 |         JsonDocument jsonDoc;
      |                      ^~~~~~~
In file included from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:28,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.h:13,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\webfunctions.cpp:1:
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:320:3: note: declared protected here
  320 |   ~JsonDocument() {}
      |   ^
[... 32 lines cut ...]
In file included from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:28,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\HeishaMon.ino:34:
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:308:3: note: declared protected here
  308 |   JsonDocument() : _pool(0, 0) {
      |   ^~~~~~~~~~~~
C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\HeishaMon.ino:275:22: error: 'ArduinoJson6194_F1::JsonDocument::~JsonDocument()' is protected within this context
  275 |         JsonDocument jsonDoc;
      |                      ^~~~~~~
In file included from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:28,
                 from c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\HeishaMon.ino:34:
c:\Users\user\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:320:3: note: declared protected here
  320 |   ~JsonDocument() {}
      |   ^

exit status 1

Compilation error: 'ArduinoJson6194_F1::JsonDocument::JsonDocument()' is protected within this context`

The JsonDocument class constructor and destructor are marked as protected, meaning they can't be directly instantiated or destroyed by your code. Who can fix it?

geduxas, 2024-09-09:
You try to compile it or modify?
For compile refere to .github workflow.. there is correct compilation schema with all required flags and settings..

IgorYbema, 2024-09-09:
You aren't using arduinojson v7 (https://arduinojson.org/v7/)

Maystero14, 2024-09-09:
Thx [IgorYbema](https://github.com/IgorYbema) is working. But now I have:
`C:\Users\user\Downloads\HeishaMon-3.8\HeishaMon\rules.cpp:76:43: error: 'MEMPOOL_ADDRESS' was not declared in this scope
   76 | unsigned char *mempool = (unsigned char *)MEMPOOL_ADDRESS;
      |                                           ^~~~~~~~~~~~~~~

exit status 1

Compilation error: 'MEMPOOL_ADDRESS' was not declared in this scope`

IgorYbema, 2024-09-09:
You really need to look at the main.yml file. It has all your answers.
Mainly you need to set MMU option correctly,
[image]

But also cpu speed to 160mhz and SSL to basic

IgorYbema, 2024-09-09:
The file is here: https://github.com/IgorYbema/HeishaMon/blob/main/.github/workflows/main.yml

But if you are a beginner that is probably the reason why you don't understand it.

Why would you like to build it yourself if you are a beginner?

Maystero14, 2024-09-09:
Compilation completed successfully. Thank you for your help.

geduxas, 2024-09-10:
@Maystero14 if your issue is resolved, please don't forget to close it.

closed 2024-09-10 by Maystero14