heishamon/HeishaMon #525: Build a larger community for HeishaMon (i.e. english forum)

mi-hol, 2024-09-11:
Open Source Projects with an engaged community are more successful in the long run.

Therefore facilitating communication is one key ingredient for building a community.

Currently only a Slack channel is mentioned in project ReadMe but a forum seems to exist for dutch users of HeishaMon.

Slack is a commercial offering in need to earn money and has therefore restrictions like hiding content, which makes it a bad fit from my view.
[image]

May I ask what is currently seen as the best forum for english speaking users of HeishaMon?

IgorYbema, 2024-09-11:
The provided slack link is for english mainly. there are some subchannels for local languages.
But it is active lately

HighlyCompressedAir, 2024-09-11:
Is it active? I can only see a few messages that are 2 months old and everything else is hidden. Are there some other channels I can't see?
Slack is pretty useless anyway, I think Github is fine for communication.

IgorYbema, 2024-09-11:
I meant inactive

HighlyCompressedAir, 2024-09-11:
Right. Github page could use a clean up though, having so many open issues may signal that the project is extremely buggy, when in reality most of those issues are long resolved or not really issues, just questions and already answered.

HighlyCompressedAir, 2024-09-11:
I can go over them in detail if you'd like to. I don't like it when issues are being closed just because they're old.

mi-hol, 2024-09-11:
@Egyras @geduxas instead of acting quickly, may I suggest to continue the conversation and agree on actions and their owner(s) ?

@HighlyCompressedAir I share many of your observations and comments.

My observations are:

1.  There seem to be a number of very active developers
2. There seems to at least an active supporter for end users
3. There seems to be no real project team with agreed roles and responsibilities

@Egyras @IgorYbema @geduxas @MiG-41 
Could you please comment on these observations as again, I lack insight into the project yet but was an IT product owner and project manager during my professional career

mi-hol, 2024-09-12:
**Proposed actions:**

- [ ]  how about [enabling GH discussions feature for this repo](https://docs.github.com/en/discussions/quickstart) as mentioned [here](https://github.com/Egyras/HeishaMon/issues/356#issuecomment-2346746619)

mi-hol, 2024-09-28:
@Egyras @IgorYbema shall I understand no comment in 2 weeks as no interest to offload work from you?

Egyras, 2024-09-28:
I doubt that there will be more chat in GH comapring with slack :)

geduxas, 2024-09-28:
Please remember that heishamon is only gateway (bridge) between automation/integration and HeatPump.. there is nothing interesting in that.. it's just hardware which opens endless possibilities.. 
Also we have slack.. and it's quiet enough...

mi-hol, 2024-09-28:
> I doubt that there will be more chat in GH comapring with slack :)

Well currently a number of "open issues" are actually questions that can be moved to "discussions".
Handling questions can be offloaded from developers easily, giving them back time for developing new features or fixing bugs.

Pre-requisite is a team of volunteers that are eager to take responsibilities and a project owner that trusts other volunteers.
The first part seems met.

mi-hol, 2024-10-15:
@geduxas "Add stale action to autoclose old issues and PR's #155" is a very dangerous action as it was activated without using "labels" to mark valid but long running issues for exemption from "stale" rule

geduxas, 2024-10-15:
Fill free to PR

mi-hol, 2024-10-15:
> Fill free to PR

Unfortunately changing a project setup in GH is impossible to change via PRs :(

mi-hol, 2024-10-15:
Therefore a conversation is required to build a project team with agree roles, responsibilities and permissions in GH first.
After that the repo owner will need to implement agreed action via manual activities in GH UI.
No shortcuts posssible!

geduxas, 2024-10-15:
@mi-hol what vision do you have for this project?

mi-hol, 2024-10-16:
@geduxas from my view I outlined the vision and missing setup in my comments made already.
I don't know your background regarding SW development and user support with teams, hence don't know how to respond in writing in order to get to a common understanding. 
Usually only talking 1:1 leads to such common understanding.  Are you available to talk?

geduxas, 2024-10-18:
> @geduxas from my view I outlined the vision and missing setup in my comments made already. I don't know your background regarding SW development and user support with teams, hence don't know how to [...]

Sorry I don't think i will be right person, i just regular user with let's say more advanced knowledge.. 
@Egyras is main codebase author as i know.. but now all development are made by @IgorYbema, als @CurlyMoo introduced rules (scripting capabilities) to heishamon..

So practically that's it. 

I don't understand what you're trying to do, main idea of heishamon is just make bridge between "something" and heatpump. Yes it could go beyond

There is a lot of tools/automations/script's and etc to control heatpump.. most of users i think comes here from those tools.. and not in vice versa.  So all discussion's should be there.

mi-hol, 2024-10-22:
I would just try to improve the user experience, no big change of software!

closed 2024-11-27 by github-actions[bot]