heishamon/HeishaMon #526: build:  This request has been automatically failed because it uses a deprecated version of `actions/upload-artifact: v2`.

mi-hol, 2024-09-11:
[build](https://github.com/mi-hol/HeishaMon/actions/runs/10809324599/job/29984185248#step:1:36)
This request has been automatically failed because it uses a deprecated version of `actions/upload-artifact: v2`. Learn more: https://github.blog/changelog/2024-02-13-deprecation-notice-v1-and-v2-of-the-artifact-actions/[build](https://github.com/mi-hol/HeishaMon/actions/runs/10809324599/job/29984185248#step:1:36)
This request has been automatically failed because it uses a deprecated version of `actions/upload-artifact: v2`. Learn more: https://github.blog/changelog/2024-02-13-deprecation-notice-v1-and-v2-of-the-artifact-actions/[build](https://github.com/mi-hol/HeishaMon/actions/runs/10809324599/job/29984185248#step:1:36)
This request has been automatically failed because it uses a deprecated version of `actions/upload-artifact: v2`. Learn more: https://github.blog/changelog/2024-02-13-deprecation-notice-v1-and-v2-of-the-artifact-actions/[build](https://github.com/mi-hol/HeishaMon/actions/runs/10809324599/job/29984185248#step:1:36)
This request has been automatically failed because it uses a deprecated version of `actions/upload-artifact: v2`. Learn more: https://github.blog/changelog/2024-02-13-deprecation-notice-v1-and-v2-of-the-artifact-actions/

@IgorYbema maybe you have already done such a required conversion to v4?

mi-hol, 2024-10-15:
All "official" merge" actions are affected too!

example "Merge pull request #551 from geduxas/patch-4 #156"
https://github.com/Egyras/HeishaMon/actions/runs/11234237944/job/31229634460#step:1:36

geduxas, 2024-10-15:
> All "official" merge actions are affected too!
> 
> example "Merge pull request #551 from geduxas/patch-4 #156" https://github.com/Egyras/HeishaMon/actions/runs/11234237944/job/31229634460#step:1:36

It's just needs version levelup.. no big deal, in next pr i think it will be fixed

closed 2024-11-25 by github-actions[bot]