heishamon/HeishaMon #528: RESET DEFAULT HEISHAMON 4.7

dimaregabriele, 2024-09-12:
Hi everyone. I have a heishamon 4.7 pcb board. I can no longer connect to the same wifi network set. so I have to reset it to factory settings to be able to access its hotspot. how can I reset it? inside there are two buttons: flash and reset. how should I press them? Thank you

IgorYbema, 2024-09-12:
If you are running the latest firmware just hold the boot button for more than 10 seconds
On older versions, you need to double reset the board within about 0.5 seconds.

In both cases you will see the led flickering fast. Then a reset will take you back in factory mode

IgorYbema, 2024-09-12:
Also, on older firmware there was no option to disable the hotspot. If heishamon can not connect to the wifi, and if the hotspot setting is enabled, it will eventually present the hotspot again. Even without factory reset.

dimaregabriele, 2024-09-12:
but which button should I press? reset ? for how long?

closed 2024-10-21 by github-actions[bot]