heishamon/HeishaMon #530: Feature Request: Enable Temperature Readings of optional PCB

baddivot, 2024-09-13:
Hi there,

I would like to run my Aquarea in a bivalent mode with a buffer. For this to work, I would need the optional PCB CZ-S5P for attaching a temperature sensor (it gets attached to E45)
Since Heishamon can emulate that optional PCB for outgoing controls (The heatpump controling external devices), I thought it should also be possible that it allows incoming readings and pass these to the heatpump, i.e. I could write the temperature I measure with my own hardware into a topic and heishamon sends it via E45 to the heatpump. 

Let me know what you think...I would prefer not having to buy the optional PCB for only one temperature reading.
Regards
baddi

HighlyCompressedAir, 2024-09-13:
Yes, you can do that. On top of that you can add 1wire temperature sensor directly to Heishamon to make it simple and reliable.

baddivot, 2024-09-13:
Thanks for you quick reply. If I understand you correctly, I can already emulate PCP readings? So how can I do that? I looked at the code and I found that all the setters for the opt Topics are skeletons and not implemented. I guess it is important for the HP to receive the temp through E45. Can I write directly to that register from whichever topic?

HighlyCompressedAir, 2024-09-13:
Are you sure you're looking at the latest version?
@SetBufferTemp = 30 makes the heat pump think the buffer temperature is 30 degrees. You can write to it anything you want whether as a rule directly in Heishamon or through mqtt server, Home Assistant and such. 1wire temperature sensor connected to Heishamon and a simple rule should be the easiest way, something like:

on ds18b20#2800000000000000 then
@SetBufferTemp = ds18b20#2800000000000000
end

Buffer temperature should then update automatically. If not there has to be a timer set up, I'm not sure, I haven't used 1wire yet, but it's easy either way.

HighlyCompressedAir, 2024-09-14:
Sorry, I didn't realize that the buffer tank sensor connection got moved from optional pcb to mainboard in newer pumps, but that's actually good for your case. All you have to do is connect a temperature sensor to the pump, enable optional pcb emulation and then enable buffer tank in settings. No rules or automations required.

baddivot, 2024-09-14:
Hi and thanks for following up on this. I have no problem with the buffer temperature, though. 

I'm referring to the outlet temp from the buffer which is read through connector E45 on the PCB. I attached the image of the hydraulic schematic. I highlighted the sensor in question. (PAW-A2W-TSHC [E45]. 
I also attached the schematic of the PCB. There I highlighted the actual connector E45.
I'm expecting that that temperature is usually a different one than the actual buffer temp in a bivalent setup, if the bivalent heat source is an oil or wood oven or so. So I need to read it for controlling the mixer.

[image]
[image]

BTW, if anybody here is in for a discussion of the official Panasonic schematics, like the one I attached, let me know. For starters, I don't understand why Panasonic does not put a mixer into the buffer outlet/heating inlet in a bivalent setup.

geduxas, 2024-09-14:
For mixer valve..  it's depends on setup.. i think by default it's left for installer to decide how will be driven heating equipment. 

As for Panasonic side, you can install mixing valve, and it will work as should, because there is no additional setup for mixing valve, Panasonic always thinks valve exist in system.

HighlyCompressedAir, 2024-09-14:
Ok, I get it, you're reffering to water sensor zone 1. Your issue actually may correlate with another one, I think we have to figure some stuff out regarding optional pcb sensors. The setup you want is actually described in the service manual:
[image]

As you can see it requires connecting sensors to main pcb, not optional. It looks like some zone 1 sensors on optional pcb aren't really used (or all of them?).

geduxas, 2024-09-14:
Also to take a note! Optional pcb emulation was made for H/J series HP. As for K/,L we just making presumption that it should work.. but really there is not tested enough.

closed 2024-10-21 by github-actions[bot]