heishamon/HeishaMon #532: SetZ1RoomTemp not working

bagbag, 2024-09-13:
My goal is to control the heat pump by using existing zigbee temperature sensors and use SetZ1RoomTemp. I guess I'm missing some configuration, but I can't get it working.

My settings:
[image]
I also tried using `External thermostat`, but that doesn't work either.

Heat pump shows either -78°C with `Thermistor` or an - and reports an H43 Error with `External thermostat` (which does make sense though).

How is `SetZ1RoomTemp` supposed to work?

Thanks!

IgorYbema, 2024-09-14:
Did you turn on 'optional pcb emulation' and turn it on on the panasonic controller also (to have the optional pcb active)?

bagbag, 2024-09-14:
Yes, other optional PCB features work flawless. I also enabled zone 2 on my heat pump to test if setting temp works there, which it does.

IgorYbema, 2024-09-14:
Can you paste your JSON data? http://heishamon.local/json

bagbag, 2024-09-14:
Of course. Directly after sending the command, I've seen that `Z1_Temp` (TOP56) was the correct value, but a second after that, it was back to -78°C.

```
{"heatpump":[{"Topic":"TOP0","Name":"Heatpump_State","Value":0,"Description":"Off"},{"Topic":"TOP1","Name":"Pump_Flow","Value":0.13,"Description":"l/min"},{"Topic":"TOP2","Name":"Force_DHW_State","Value":0,"Description":"Disabled"},{"Topic":"TOP3","Name":"Quiet_Mode_Schedule","Value":0,"Description":"Disabled"},{"Topic":"TOP4","Name":"Operating_Mode_State","Value":0,"Description":"Heat"},{"Topic":"TOP5","Name":"Main_Inlet_Temp","Value":22.50,"Description":"°C"},{"Topic":"TOP6","Name":"Main_Outlet_Temp","Value":22.50,"Description":"°C"},{"Topic":"TOP7","Name":"Main_Target_Temp","Value":28,"Description":"°C"},{"Topic":"TOP8","Name":"Compressor_Freq","Value":0,"Description":"Hz"},{"Topic":"TOP9","Name":"DHW_Target_Temp","Value":50,"Description":"°C"},{"Topic":"TOP10","Name":"DHW_Temp","Value":43,"Description":"°C"},{"Topic":"TOP11","Name":"Operations_Hours","Value":89,"Description":"hours"},{"Topic":"TOP12","Name":"Operations_Counter","Value":39,"Description":"count"},{"Topic":"TOP13","Name":"Main_Schedule_State","Value":0,"Description":"Disabled"},{"Topic":"TOP14","Name":"Outside_Temp","Value":10,"Description":"°C"},{"Topic":"TOP15","Name":"Heat_Power_Production","Value":-200,"Description":"Watt"},{"Topic":"TOP16","Name":"Heat_Power_Consumption","Value":-200,"Description":"Watt"},{"Topic":"TOP17","Name":"Powerful_Mode_Time","Value":0,"Description":"Off"},{"Topic":"TOP18","Name":"Quiet_Mode_Level","Value":0,"Description":"Off"},{"Topic":"TOP19","Name":"Holiday_Mode_State","Value":0,"Description":"Off"},{"Topic":"TOP20","Name":"ThreeWay_Valve_State","Value":0,"Description":"Room"},{"Topic":"TOP21","Name":"Outside_Pipe_Temp","Value":10,"Description":"°C"},{"Topic":"TOP22","Name":"DHW_Heat_Delta","Value":-10,"Description":"K"},{"Topic":"TOP23","Name":"Heat_Delta","Value":3,"Description":"K"},{"Topic":"TOP24","Name":"Cool_Delta","Value":3,"Description":"K"},{"Topic":"TOP25","Name":"DHW_Holiday_Shift_Temp","Value":-15,"Description":"K"},{"Topic":"TOP26","Name":"Defrosting_State","Value":0,"Description":"Disabled"},{"Topic":"TOP27","Name":"Z1_Heat_Request_Temp","Value":20,"Description":"°C"},{"Topic":"TOP28","Name":"Z1_Cool_Request_Temp","Value":28,"Description":"°C"},{"Topic":"TOP29","Name":"Z1_Heat_Curve_Target_High_Temp","Value":28,"Description":"°C"},{"Topic":"TOP30","Name":"Z1_Heat_Curve_Target_Low_Temp","Value":28,"Description":"°C"},{"Topic":"TOP31","Name":"Z1_Heat_Curve_Outside_High_Temp","Value":15,"Description":"°C"},{"Topic":"TOP32","Name":"Z1_Heat_Curve_Outside_Low_Temp","Value":-10,"Description":"°C"},{"Topic":"TOP33","Name":"Room_Thermostat_Temp","Value":21,"Description":"°C"},{"Topic":"TOP34","Name":"Z2_Heat_Request_Temp","Value":20,"Description":"°C"},{"Topic":"TOP35","Name":"Z2_Cool_Request_Temp","Value":28,"Description":"°C"},{"Topic":"TOP36","Name":"Z1_Water_Temp","Value":-78,"Description":"°C"},{"Topic":"TOP37","Name":"Z2_Water_Temp","Value":-78,"Description":"°C"},{"Topic":"TOP38","Name":"Cool_Power_Production","Value":-200,"Description":"Watt"},{"Topic":"TOP39","Name":"Cool_Power_Consumption","Value":-200,"Description":"Watt"},{"Topic":"TOP40","Name":"DHW_Power_Production","Value":-200,"Description":"Watt"},{"Topic":"TOP41","Name":"DHW_Power_Consumption","Value":-200,"Description":"Watt"},{"Topic":"TOP42","Name":"Z1_Water_Target_Temp","Value":28,"Description":"°C"},{"Topic":"TOP43","Name":"Z2_Water_Target_Temp","Value":28,"Description":"°C"},{"Topic":"TOP44","Name":"Error","Value":"No error","Description":"Error"},{"Topic":"TOP45","Name":"Room_Holiday_Shift_Temp","Value":-15,"Description":"K"},{"Topic":"TOP46","Name":"Buffer_Temp","Value":-78,"Description":"°C"},{"Topic":"TOP47","Name":"Solar_Temp","Value":-78,"Description":"°C"},{"Topic":"TOP48","Name":"Pool_Temp","Value":-78,"Description":"°C"},{"Topic":"TOP49","Name":"Main_Hex_Outlet_Temp","Value":21,"Description":"°C"},{"Topic":"TOP50","Name":"Discharge_Temp","Value":12,"Description":"°C"},{"Topic":"TOP51","Name":"Inside_Pipe_Temp","Value":13,"Description":"°C"},{"Topic":"TOP52","Name":"Defrost_Temp","Value":-128,"Description":"°C"},{"Topic":"TOP53","Name":"Eva_Outlet_Temp","Value":11,"Description":"°C"},{"Topic":"TOP54","Name":"Bypass_Outlet_Temp","Value":-128,"Description":"°C"},{"Topic":"TOP55","Name":"Ipm_Temp","Value":-128,"Description":"°C"},{"Topic":"TOP56","Name":"Z1_Temp","Value":-78,"Description":"°C"},{"Topic":"TOP57","Name":"Z2_Temp","Value":-128,"Description":"°C"},{"Topic":"TOP58","Name":"DHW_Heater_State","Value":0,"Description":"Blocked"},{"Topic":"TOP59","Name":"Room_Heater_State","Value":0,"Description":"Blocked"},{"Topic":"TOP60","Name":"Internal_Heater_State","Value":0,"Description":"Inactive"},{"Topic":"TOP61","Name":"External_Heater_State","Value":0,"Description":"Inactive"},{"Topic":"TOP62","Name":"Fan1_Motor_Speed","Value":0,"Description":"r/min"},{"Topic":"TOP63","Name":"Fan2_Motor_Speed","Value":0,"Description":"r/min"},{"Topic":"TOP64","Name":"High_Pressure","Value":5.8,"Description":"Kgf/cm2"},{"Topic":"TOP65","Name":"Pump_Speed","Value":0,"Description":"r/min"},{"Topic":"TOP66","Name":"Low_Pressure","Value":0,"Description":"Kgf/cm2"},{"Topic":"TOP67","Name":"Compressor_Current","Value":0.0,"Description":"Ampere"},{"Topic":"TOP68","Name":"Force_Heater_State","Value":0,"Description":"Inactive"},{"Topic":"TOP69","Name":"Sterilization_State","Value":0,"D [... cut ...] ame":"Z1_Water_Pump","Value":"0","Description":"On"},{"Topic":"OPT1","Name":"Z1_Mixing_Valve","Value":"0","Description":"Off"},{"Topic":"OPT2","Name":"Z2_Water_Pump","Value":"0","Description":"Off"},{"Topic":"OPT3","Name":"Z2_Mixing_Valve","Value":"0","Description":"Nothing"},{"Topic":"OPT4","Name":"Pool_Water_Pump","Value":"0","Description":"On"},{"Topic":"OPT5","Name":"Solar_Water_Pump","Value":"0","Description":"Off"},{"Topic":"OPT6","Name":"Alarm_State","Value":"0","Description":"Off"}]}
```

IgorYbema, 2024-09-14:
Your zone1 water temp also incorrect. Maybe that is why the heatpump isn't storing your zone temp info.
So try to 'SetZ1WaterTemp' also

But my info about zones configs is a bit rusty. Not using it for a long time now.

Have to go out for the day now. I hope you can find it and maybe some else jumps in to help.

bagbag, 2024-09-14:
Hm, I'll try that later. But I don't use multiple zones (only used z2 for testing).

But as I only have a single zone directly connected to the HP, I don't have a water temperature sensor and simply used "Water temperature" for z1. Will the controlling work, if I just send any reasonable value instead of the real one? I guess it will cause problems.

geduxas, 2024-09-14:
For 2 zones, you must have water outlet temperature sensors for each of zones.. without that naturally hetpump will show error.

bagbag, 2024-09-14:
Yes that does make sense. But as I only have 1 zone, 2nd is deactivated (as seen in the screenshot).

geduxas, 2024-09-14:
Yes but you have enabled 2 zones heating..  for that it's required optional pcb and 2 temp sensors for minimum

geduxas, 2024-09-14:
Also note that both sensors becomes HP heating temperature target points, it will ignore main outlet sensor

HighlyCompressedAir, 2024-09-14:
He wrote he enabled 2 zones temporarily for testing only.
My guess would be that in 1 zone system the pump uses mainboard connector for room sensor and tries to read actual value. If you enable 2 zones then it uses the non-existent optional pcb connectors and only then you can use @SetZ1RoomTemp. I can't test it right now, please check if my theory is correct.

geduxas, 2024-09-14:
> He wrote he enabled 2 zones temporarily for testing only. My guess would be that in 1 zone system the pump uses mainboard connector for room sensor and tries to read actual value. If you enable 2 [...]

No, after you enable multiple zones, you can disable zones separately (leave only 1 or 2 or both) it will not go back to one zone operation mode, you must disable multiple zone mode at all to control without extension board

bagbag, 2024-09-14:
@HighlyCompressedAir having 2 zones enabled also didn't allow me to set z1 temp.

@geduxas I also disabled 2 zones completely (and tested in all variations I can think of). Z1 temp never worked.

While this issue is valid in itself, it is also a xy-problem. Maybe you can help me there in another way. When I have heating and dhw enabled, the compressor turns off directly after switching form dhw to heating (because inlet is still hot and z1 target outlet is 28 deg). I want to avoid that cycling, as it is the other way around (heating -> dhw), where the compressor keeps running.

HighlyCompressedAir, 2024-09-14:
I have a suspicion that zone 1 room sensor is only a dummy on the optional pcb. The heat pump always uses the one on the main board. Look here:
[image]

If it's true then it can't be emulated.

geduxas, 2024-09-14:
> I have a suspicion that zone 1 room sensor is only a dummy on the optional pcb. The heat pump always uses the one on the main board. Look here: ![bivalent](https://private-user-images.githubusercontent.com/134067260/367527957-ba3e0084-65ce-4acd-a699-8f0a8a709e90.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzMzExMDgsIm5iZiI6MTcyNjMzMDgwOCwicGF0aCI6Ii8xMzQwNjcyNjAvMzY3NTI3OTU3LWJhM2UwMDg0LTY1Y2UtNGFjZC1hNjk5LThmMGE4YTcwOWU5MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxNFQxNjIwMDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYjYxYTI2NDNjZTYzY2EzZjI0ODUxMTI4NWEzYTIxNjZiNTZmNGYyOWE5NDk3YTZlYTU5MzgyMDQ4NjU4YTFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Bo5bEcFdYy4x08_z3TmQ_25Rb0bPagP1fdQQRm6upNc)
> 
> If it's true then it can't be emulated.

You are wrong, outlet sensor is used only in simple heating mode, as soon you enable multi-zone. It will switch to optional pcb to control outlet temperature

HighlyCompressedAir, 2024-09-14:
On H and J series yes. K and L may work differently. For example there is no room thermostat 1 on optional pcb anymore, the heat pump uses the one the main board. Buffer temperature sensor was removed from optional and put on main.

HighlyCompressedAir, 2024-09-18:
I finally had access to my pump and was able to confirm it. @SetZ1RoomTemp doesn't work on K series. So either the service manual is right and it can't work or there's something wrong with the command itself, but I doubt it.

geduxas, 2024-09-18:
@HighlyCompressedAir pcb emulation was made for H/J series.. it's emulates PCB communication, nothing else. If Panasonic decided to move some headers from optional pcb to mainboard, it will naturally not accept data which is not expecting from native pcb.. also it could lead to lockout's or error's.

MiG-41, 2024-09-18:
It was made also from sniffing communication beetween H series and CZ-NS4P optional PCB. Now it is CZ-NS5P , and we don't know ,if there is something in addition....

HighlyCompressedAir, 2024-09-18:
I've tested SetPoolTemp, SetSolarTemp, SetZ1WaterTemp, SetZ2WaterTemp, SetZ2RoomTemp and they all work correctly. Another page from L series service manual:
[image]

MiG-41, 2024-09-18:
How you drive your HP ? Now ,when i use External Thermostat , TOP56 is identical to TOP36... Maybe you need to st to Intrnal thermostat or termistor , to se proper temperatur in TOP56 ?

HighlyCompressedAir, 2024-09-18:
If I set zone 1 to water or external thermostat then Z1_Temp = SetZ1WaterTemp.
If I set it to internal thermostat then Z1_Temp = Room_Thermostat_Temp.
If I set it to thermistor then Z1_Temp = -78 and I get H67 error "Abnormal External Thermistor 1".

Zone 2 works the same except SetZ2RoomTemp works correctly and there's no error.

geduxas, 2024-09-18:
If you don't have temperature sensor so it will end in error.. so only z2 temperature sensor could be emulated only

MiG-41, 2024-09-18:
So i changed for test system for termistors , and SetZ1RoomTemp (as well as SetZ2RoomTemp ) , works ok , they are influencing on TOP56 and TOP57 as they should. I have gen.H 12kW T-CAP , heishamon ver.3.62.

So it is either K specific case ( or bug) , and/or CZ-NS5P issue ( could vary from CZ-NS4P). 
Assuming , that Panasonic didn't broke it in K ( K+ CZ-NS5P works ok ) we need to sniff real communication beetween this setup to find out where could be difference.

geduxas, 2024-09-18:
@MiG-41 Panasonic party moved inputs from expansion board to main board. Z1 temperature sensor. Buffer temp sensor.. and maybe others.. so everything what is missing in new expansion board would likely not work, or will be just ignored

stundenblume, 2024-09-30:
I had a similar problem (WH-MDC05J3E5). For me this works:

**In Panasonic Menue:**
1. system setup - optional PCB: Yes
2. system setup - 1 zone system - room: room thermistor

**HeishaMon:**
1. Settings - Emulate optional PCB: yes

**Sending to MQTT topic:**
panasonic_heat_pump/commands/SetZ1RoomTemp: 22.2

**Get:**
panasonic_heat_pump/main/Z1_Temp: 22

HighlyCompressedAir, 2024-10-01:
> While this issue is valid in itself, it is also a xy-problem. Maybe you can help me there in another way. When I have heating and dhw enabled, the compressor turns off directly after switching form [...]

I forgot about your other issue. To avoid compressor shutdown you have to increase z1 target temp to match current outlet temp (or higher), then change mode from dhw to heat before dhw completes it's cycle and then after outlet temp falls below your desired target temp, undo the previous increase.

bagbag, 2024-10-02:
@HighlyCompressedAir 
Thanks, that makes sense.
Have you tested what happens when the cycle completes on itself having the Z1 target temperature above the current outlet temperature?

HighlyCompressedAir, 2024-10-02:
I just use SetCurves {"zone1":{"heat":{"target":{"high":60,"low":60}}}} and then SetCurves again with my normal curve.
You can also setup zone 2 with 60,60 curve controlled by an external thermostat. That way you increase/decrease target temp with SetExternalThermostat2State. That's probably the most reliable and 0-write way to do this.

bagbag, 2024-10-04:
@HighlyCompressedAir thanks for your tip to simply increase target z1 temp, that worked! The only thing I would like to avoid is having to set quiet mode to 3 and enable pump service mode, otherwise the compressor won't slow down enough to get below target temp. Do you have an idea for that too?

That's my first node-red flow I did, so I guess it can be made better, but well.
[image]

HighlyCompressedAir, 2024-10-04:
My pump runs at max speed setting (26l/min), not delta T in heat mode, so I don't have that problem. I don't need quiet mode or demand control, the temperature falls back down pretty quickly to where it was before. I just set a 3 minute timer after mode switch to adjust the curve to keep it simple. Compressor frequency goes up dramatically for a while, but it doesn't really matter, altough I may use quiet mode in the future to avoid that.
I don't use NodeRed, so won't help you there.

stundenblume, 2024-11-06:
@HighlyCompressedAir 
> I just use SetCurves {"zone1":{"heat":{"target":{"high":60,"low":60}}}} and then SetCurves again with my normal curve.
> You can also setup zone 2 with 60,60 curve controlled by an external [...]

I really liked your idea to trick the HP by moving the Z1 curve. It seams to work well (today first DHW with the rule working).
I also tried to use the Z2 approach, but I don't own the optional PCB and setting "SetZ2RoomTemp" didn't work. Using "SetZ1RoomTemp" without problem. Also I get H43 and H74 errors when using both zones and sending to both roomTemp topics.

Simplified Rule:
~~~
on System#Boot then
  $version = '241106-17';
  #boot = 1;
  #flushTemp = 55;
  #flushT = #flushTemp;
  #flushTime = 600;
 
  #Z1HCTHT = 26;
  #Z1HCTLT = 26;
 
  setTimer(1,10);
end
 
on timer=1 then
  #boot = 0;
end
 
on timer=2 then
  setZ1HCTT(#Z1HCTHT, #Z1HCTLT);
end
 
on setZ1HCTT($highT, $lowT) then
  @SetCurves = concat('{zone1:{heat:{target:{high:', $highT,',low: ', $lowT,'}}}}');
end
 
on @ThreeWay_Valve_State then
  if #boot == 0 then
    if @ThreeWay_Valve_State == 1 then
      setZ1HCTT(#flushTemp, #flushT);
    end
    if @ThreeWay_Valve_State == 0 then
      setTimer(2, #flushTime);
    end
  end
end
~~~

closed 2024-12-12 by github-actions[bot]