heishamon/HeishaMon #533: Looking for ideas how to use heat pump thermostat with external temperature

jaco, 2024-09-16:
I've encountered a challenge that I hope someone here might help me solve :)

The heat pump's internal thermostat is located in the boiler room, where the temperature is consistently around 33°C (91°F). Unfortunately, relocating this sensor into the living spaces is not an option for me. I want to consider temperatures from several rooms in my home, not just a single location, to achieve a more balanced and efficient heating control.

I can create virtual thermostat in HA to control heat pump (altering heat curve, enabling powerful mode?) but I am afraid that hysteresis a little to dumb and not efficient :) 

Ideal solution will be to somehow overwrite panasonic's thermostat temperature from HA.

geduxas, 2024-09-16:
Look for node-red integration, it can drive +5/-5 curve shift to desired temperature.. as for input you could make your own logic (for example pick lowest room temperature) and pass it as target temperature..

jaco, 2024-09-16:
I don't have any problems with automations but with that "own logic" :) I understand hysteresis and that I can do but anything else is foreign…

How would you alter heat curve if you want to make, for example -7°C change to room temperature?

Or maybe there is a way to connect external room temp sensor directly to heap pump and emulate that sensor? :)

geduxas, 2024-09-16:
> I don't have any problems with automations but with that "own logic" :) I understand hysteresis and that I can do but anything else is foreign…
>
> How would you alter heat curve if you want to make, [...]

Yes, use room thermistor, i think it's possible to emulate via optional pcb.. but really try node-red.. it's awesome project.. and with right compensation curve, +3/-3 range is more than enough 

It can be roon in HA, also it's possible to control from HA

https://github.com/edterbak/NodeRed_Heishamon_control

closed 2024-10-27 by github-actions[bot]