heishamon/HeishaMon #534: Factory reset not working

westerner0101, 2024-09-17:
Hi, I have board marked "Heishamon OT v1.3".  I mistakenly uploaded the wrong firmware ('large', when it should have been 'small'), and now the Heishamon no longer responds.  I am now trying to factory reset the board by pressing the reset button twice in 0.5 seconds but the blue LED just flashes once each time I press the reset button (I have connected wires to 5V and GND, via a TTL module).  It never blinks rapidly, and the default hotpot never appears.  There is a single green LED constantly lit on the board.

Can anyone please tell me exactly what I should do to factory reset the board?  Or if anyone knows a Youtube video that shows the procedure, it would be a great help.

Thank you for your time.

geduxas, 2024-09-17:
Please read #502

westerner0101, 2024-09-17:
Thanks, but I still have no success with that method.  It says "When connected you put the heishamon in firmware update mode by holding the boot button while resetting it. You can release the boot button after the reset".

I don't have a boot button, only reset and flash.  If I hold down the flash button the press (once?) the reset button, nothing happens - I can't connect with this tool: https://espressif.github.io/esptool-js/ - "failed to connect with the device".  What baudrate is recommended?

Thank you.

IgorYbema, 2024-09-17:
Please join https://github.com/Egyras/HeishaMon/issues/535

qbakk, 2024-09-17:
everything looks exactly the same for me and it doesn't work

closed 2024-10-27 by github-actions[bot]