heishamon/HeishaMon #535: Flash by TTL (v4.7)

qbakk, 2024-09-17:
Hi, I made a mistake and I tried flash board v4.7 by "model-type-large" 3.8 firmware.

I wan't to fix it and i try to use
CP2102 Board and I have problem with connection on website: https://espressif.github.io/esptool-js/

All my try end that:
[image]

(on the screen I tried use the FT232RL from my friend)

what i do wrong?

IgorYbema, 2024-09-17:
Make sure tx and rx are placed correctly AND press-and-hold flash button while resetting the board to get the board into manual firmware upload mode.

IgorYbema, 2024-09-17:
Also make sure you upload the heishamon binary to address 0x0000 and not 0x1000

qbakk, 2024-09-17:
I make sure the tx and rx i correctly (I saw the post with similar problem). When I hold the flash button nothing happens always green led is on. When I click reset button blue led blink (on esp)

westerner0101, 2024-09-17:
> Make sure tx and rx are placed correctly AND press-and-hold flash button while resetting the board to get the board into manual firmware upload mode.

"while resetting the board" - can you please clarify this procedure? Do you mean holding the reset button, or something else? 

Also, when using a TTL module with Dupont wires, is it TX-TX & RX-RX or TX-RX & RX-TX?

Thanks again.

IgorYbema, 2024-09-17:
1) press-and-hold flash button
2) press reset
3) release flash button after a second

This will bring it into flash mode.

The wiring should be TX->RX and vice versa. But if that doesn't work just try it the other way arround  (maybe the markings on your dupont wire is wrong).

qbakk, 2024-09-17:
for me work's all fine!

westerner0101, 2024-09-17:
> 1. press-and-hold flash button
> 2. press reset
> 3. release flash button after a second
> 
> This will bring it into flash mode.
> 
> The wiring should be TX->RX and vice versa. But if that doesn't work just try it the other way arround (maybe the markings on your dupont wire is wrong).

Thank you very much for your help, it's all working again now 👍🏻 (it was TX-RX for me)

Egyras, 2024-09-19:
resolved

closed 2024-09-19 by Egyras