heishamon/HeishaMon #538: Restart rules or reboot with a rule?

HighlyCompressedAir, 2024-09-19:
I'd like Heishamon to restart all the rules when I manually turn off the heat pump. Or reboot maybe? Right now when I turn off my pump all timers are still going, variables end up in the wrong state and so on. Is there a way to do this?

McMagellan, 2024-09-20:
If by switching off you mean the button on the remote control (with a green LED), you can easily recognize this in Heishamon using the `TOP0 Heatpump_State` parameter and react to it in the rules. 
But if you switch off the power to the WP (Heishamon still has power), Heishamon will only notice it if you set up a watchdog Rule that changes a parameter cyclically and if this change is no longer reflected in the rules, the WP is gone. As far as I know, resetting Heishamon or the WP from Rules isn't possible, but it can be done using a url command. (downstream SmartHome).

For example: Trigger Rule check Heatpump State:

```
on @Heatpump_State then
   if @Heatpump_State == 0 then
      settimer(31,0);
      settimer(34,0);
      #TestBonusLangsam = 0;
      settimer(32,0);
      #TestBonusQuick = 0;
      settimer(33,0);
      #TestBonusAbbau = 0;
      settimer(35,0);
      settimer(36,0);
      #Bonus = 0;
      #Quicky = 0;
      HKUpdate();
      settimer(40,15);
   end
end
```

HighlyCompressedAir, 2024-09-20:
I meant turn off by pushing the remocon button, yes. So if I have timer=31 already running then doing settimer(31,0) cancels it? And it won't trigger?

McMagellan, 2024-09-20:
Here is an example of how something like this can be realized.
1 A function (CheckPanaRun) checks the current status of the TOP0.
1a If the WP is switched on, timer 20 is started after 2 seconds, which then calls itself endlessly every 60 seconds.
1b If the WP was switched off, timer 20 is stopped.
2 The function (CheckPanaRun) is called when the rule starts and every time the value of TOP0 (on/off) changes. 

Sometimes the TOP variables are not yet updated after starting the Rulesset, which is why Timer1 can be placed between the System#Boot and a TOP query.

3 The `#PanaRun` variable can be used as a status signal in the rules.

```
on System#Boot then
   #PanaRun = 0;
   settimer(1,15);
end

on timer=1 then
   print("Check after start");
   CheckPanaRun();
end

on @Heatpump_State then
   print("Check when state was changed");
   CheckPanaRun();
end

on CheckPanaRun then
   if @Heatpump_State == 1 then
      if #PanaRun == 0 then
         #PanaRun = 1;
         settimer(20,2);
      end
   else
      if #PanaRun == 1 then
         #PanaRun = 0;
         settimer(20,0);
      end
   end
end

on timer=20 then
   print("Minute Loop");
   settimer(20,60);
end
```

The print- command is Beta- Release. Do not use it excessiv.

HighlyCompressedAir, 2024-09-21:
I need some clarification on how timers work.
Setting timer time to 0 cancels existing instance of that timer so it won't trigger. So if timer=x is running and has 90 seconds to go and settimer(x,0) is called then timer=1 cancels and won't trigger, correct? I thought that setting timer time to 0 just triggers it immediately.

How about if a timer=x is already running and has 90 seconds to trigger, but then another settimer(x,120) is set?
1. Does the first instance cancel and another appear with timer set to 120 seconds?
2. Does the time add up and now it's on 90+120 seconds?
3. Are there two instances of the same timer running at the same time and one triggers after 90 seconds and the other after 120 seconds?

McMagellan, 2024-09-21:
A timer only exists once, which is why multiple calls are not possible.
It can be done with settimer(xx,yy); can be preset to a delay time yy in seconds.
The system counts down the timer time by 1 every second.
When switching from 1 to 0 the timer is executed.
Each followed time setting overwrites the current value.
settimer(xx,0); stops the timer immediately without execution.
The currently running time value cannot be read.
If all timers have expired and no trigger parameter has been set, the program is over.
A program loop must therefore include the re-invocation of its own timer.

closed 2024-10-29 by github-actions[bot]