heishamon/HeishaMon #539: Spontaneous idea for a "commands page" in Heishamon

McMagellan, 2024-09-21:
Last week I wasn't at home, but I was able to connect to Heishamon via VPN from my cell phone.
That's when the desire arose to be able to change parameters directly in Heishamon.
There are three ways to change parameters:
1 return channel MQTT connection from the SmartHome
2 URL Commando
3 From a running rule

I then came up with the following:
The main menu is expanded to include the “commands” menu item, which leads to a page with a small input mask. This could be structured in the style like “settings”. Here, in the drop-down field, a selection of the categories in which the parameters are grouped is made. The desired parameter can then be selected from another drop-down field. Only the parameters that can be changed are listed. An info field now shows the valid range in a 3-line field as text. A new value can now be set and transferred to the HP with “send”.

[image]

This makes it easy to safely change parameters. It would enhance the professional character of the Gateway Heishamon. 
The reaction can then be followed in the console. 
Last but not least, this page can be used as a lexicon if you program in rules or via url and look for the correct syntax of a command and then use it with copy/paste.

What do you think about it?

HighlyCompressedAir, 2024-09-21:
Looks overcomplicated . I'd just add another column on the main page called "Set Parameter" and then a dropdown menu or an empty field next to a parameter than can be changed.

Topic	Name				Value	Description	Set Parameter
TOP0	Heatpump_State		1		On			(dropdown menu with 0 - OFF and 1 - ON options) 	(Set button)
TOP9	DHW_Target_Temp	50		°C			(empty field to write in a value)	40-65 (value range)	Set button

That would definitely move Heishamon closer to replacing CZ-TAW1.

stundenblume, 2024-10-01:
Here is also a very nice example of a simple GUI.

https://github.com/fredlcore/BSB-LAN

geduxas, 2024-10-01:
Yes i thought about same feature, i think it's really possible to implement, how about that small feature @IgorYbema ? Will it fit in memory?

stundenblume, 2024-10-02:
For me it don't have to look fancy, but [WLED](https://kno.wled.ge/) is the best looking, simple and most user friendly (desktop & mobile supported) UI I have ever seen on an ESP8266. Just to get an idea what would be possible.

McMagellan, 2024-10-02:
Unfortunately, none of the developers have commented on this. 
I would prefer a very simple version without graphic elements. Basically it's a small word processor. The texts could all be in a data array. On the “Settings” page there are already drop-down fields that are filled individually in the case of WiFi or have a large number of entries when selecting the time zone.

In addition to the existing sensor ports, Heishamon V5 adds actuators. It is advantageous to be able to switch these manually with Heishamon. This would also be a good solution for the missing alias text support.

geduxas, 2024-11-02:
This should be left open. Or move to discussion

McMagellan, 2024-11-02:
Sorry, I was too creative.

geduxas, 2024-11-02:
It's ok, i think it's possible to implement, but Ned's time and attention :) @IgorYbema is gone somewhere, didn't hear anything from him for a while.

McMagellan, 2024-11-02:
OK.
There is also an error in memory management that always causes Heishamon to crash if a webserver connection (console) exists for a longer time and then sometimes the rules are not automatically reloaded.

My CoPilot ruleset is now 12KB in size with 600 lines in 33 rules and runs absolutely reliably and correctly on Heishamon v4 when there is no webserver connection.

I have now ordered Heishamon v5 because of the ESP32 environment and to use it as a proxy. 

Do you provide a link to the V5- documentation?

geduxas, 2024-11-02:
It's @IgorYbema (TheHogNL) hardware, yes esp32 is more powerful and have more memory, it could work more stable, as for memory, i think it's not heishamon problem.. it's itself esp8266 stack, or web server library which is used in project. So it will be hard to figure out.

IgorYbema, 2024-11-02:
> It's @IgorYbema (TheHogNL) hardware, yes esp32 is more powerful and have more memory, it could work more stable, as for memory, i think it's not heishamon problem.. it's itself esp8266 stack, or web server library which is used in project. So it will be hard to figure out.

still around but this time of the year there are always a lot of shipping that needs be be done so no time for coding or reading on issue. Luckaly we have a rising community of good people :)

McMagellan, 2024-12-03:
throw it away. Idea for the trash.

geduxas, 2024-12-03:
Move it to discussion, i think idea is great, but it's need time.

geduxas, 2025-01-04:
@IgorYbema have you time to look for this idea?

closed 2025-02-10 by github-actions[bot]