heishamon/HeishaMon #540: Aquaera doesn't get any orders from Heishamon

hubsi5, 2024-09-21:
I have the Panasonic INET module and the large Heishamon running in parallel.

I get all the data.

I can control via the Panasonic Cloud.

Heishamon reliably provides me with all the data.

But now I want to send commands with Heishamon and unfortunately that's just a game of chance.

Sometimes it works, sometimes it doesn't.

I already have the Panasonic module and Heishamon directly on the LAN.

Regardless of whether I send commands via MQTT or HTTP, Heishamon only passes them on every now and then.

But in the Heishamonlog I see that the command arrives.
[image]

IgorYbema, 2024-09-21:
You said you have the INET module (cz-taw1) parallel. This will not work if heishamon is also running. You need to connect this cz-taw1 to the proxy port of the heishamon and not directly to the heatpump

hubsi5, 2024-09-21:
Oh thank you.
I guess I had a blackout then
Thanks, it works now!

closed 2024-09-21 by hubsi5