heishamon/HeishaMon #542: How to do a firmware update

klausj1, 2024-09-22:
The firmware update description in the readme is pretty short ;)

Just to be sure: To get the correct binaries (currently 3.8), download `source.zip` from [here](https://github.com/Egyras/HeishaMon/releases/tag/v3.8), and from `HeishaMon-3.8\binaries\model-type-small` (I have the small board, no LAN), take `HeishaMon.ino.d1-v3.8.`bin.

Correct?

A small update of the documentation would be great ;)

IgorYbema, 2024-09-22:
No need to download the complete sources. You can easily download the binary from https://github.com/Egyras/HeishaMon/tree/master/binaries

Also take the .md5 file and read the content. Paste that md5 text into the md5 input field of the heishamon update page.

klausj1, 2024-09-22:
Lightning-fast, as usual 😊 Thanks! A link to this page in the docu would be great.

Stueckgold, 2024-09-25:
I try to flash from 3.2.3 to 3.8 or any other binary:
I always get errors
"Update failed! Please try again...ERROR[7]: MD5 Failed"
or without md5 it´s
"Update failed! Please try again...ERROR[10]: Magic byte is wrong, not 0xE9"

small version

IgorYbema, 2024-09-25:
@Stueckgold you are probably not downloading the files correctly. Downloading the HTML content instead of the real/raw binary file.
And try to upload without a second tab open to heishamon (no console log running)

Stueckgold, 2024-09-29:
You was right, no right-click save to...

closed 2024-10-22 by Egyras