heishamon/HeishaMon #544: wifi does not work and led shows nothing

kirchma, 2024-09-25:
I have the large version of the HeishaMon v5 PCB (date 05-2024). After I was initially able to set everything up, the WLAN stopped working and I could no longer access it. I uploaded the firmware HeishaMon.ino.esp32-v3.5.bin (model-type-large) with https://espressif.github.io/esptool-js/ using a USB-TTL cable. The upload goes through but nothing happens afterwards. No wifi signal, the status-led stays off all the time.  Can anyone help me?

geduxas, 2024-09-25:
#535

kirchma, 2024-09-26:
I have tried suggestions there but without success. 
I have activated the flash mode, changed the binary address to 0x0000 and also tried TX and RX in both combinations. The ESP tool always runs through and outputs “Hash of data verified” at the end. During the whole process the status LED never lights up and the wifi is not visible. The status LED seems strange to me, any other tips?

kirchma, 2024-09-30:
I have tried everything but the status led never lights up. Even when I use the LAN module, I can't see HeishaMon on the network.
Is this simply a hardware defect? I mean the LED should light up in some color?

geduxas, 2024-09-30:
You're doing something wrong. Does flash successful? Maybe try to erase whole memory and try from blank

IgorYbema, 2024-09-30:
Can you try using the usb-c and not TTL to upload the firmware?

IgorYbema, 2024-09-30:
And are your sure you are uploading the binary file and not a 'download' of the github webpage? (which some people do by mistake)

kirchma, 2024-09-30:
1. I hold the boot button, press the reset button for 1 second and then release the boot button. Ι should be in flash mode now?
2. I only used the usb-c connection to upload this [file](https://github.com/Egyras/HeishaMon/blob/master/binaries/model-type-large/HeishaMon.ino.esp32-v3.8.bin). 
3. result see screenshots. Then I click disconnect and press the reset button once. Meanwhile, the status led is not lit, which is still unsettling.

[image]

IgorYbema, 2024-09-30:
You are uploading the OTA file which only works when you use the heishamon webinterface. You are now overwrinting the boot, partitions etc. 
Let me look for the correct file

IgorYbema, 2024-09-30:
Use the binary from this zip to upload to address 0x000 (it includes the partions, bootloader etc)
[HeishaMon.ino.esp32-v3.8.complete.zip](https://github.com/user-attachments/files/17195468/HeishaMon.ino.esp32-v3.8.complete.zip)

kirchma, 2024-10-03:
Looks like the problem is once again sitting in front of the monitor ... thank you very much. 
I have an additional question, as soon as I activate “use s0 kWh metering” under settings (GPIO 12 & 14) and reboot heishamon, the lan connection is lost and the status led lights up red. From here on the factory reset (press boot button for 10 sec) does not work either. Is there a bug or am I doing something wrong again?

firen456, 2024-10-14:
hello,

I'm also using the large version and managed to brick it somehow...
@IgorYbema I used your 3.8 binary with partitions and bootloader and flashed it via USB-C. Flashing was fine according to esptool-js...
But I cant get it working. Neither via LAN-Module oder Wifi Hotspot. There's no connection.
Do u have any idea?

IgorYbema, 2024-11-14:
> hello,
>
> I'm also using the large version and managed to brick it somehow... @IgorYbema I used your 3.8 binary with partitions and bootloader and flashed it via USB-C. Flashing was fine according to [...]

You need https://github.com/IgorYbema/HeishaMon/blob/main/binaries/model-type-large/HeishaMon.ino.esp32-v3.8.complete.bin to flash directly (to flash address 0x0000)

closed 2024-12-21 by github-actions[bot]